chipKIT32/chipKIT-core #272: FLASH Library: Impliment

JacobChrist, 2016-09-24:
The EEPROM emulation on chipKIT is great for Ardunio compatibility but it would be nice to have our own FLASH library that has calls that more correctly allow for reading and writing to the block of flash that is reserved for EEPROM emulation.  This would be an alternative library that would instead of the EEPROM library for chipKIT only applications.  When instantiated a ram buffer could be allocated that is used for the entire length of the sketch.  The FLASH memory would then be loaded into this buffer for immediate access until a Save function was called.

bool FLASH.begin(uint32_t size): Allocate a buffer size size bytes and load FLASH into buffer.  Returns true of buffer successfully allocated.
void FLASH.clear(): Clear the FLASH buffer.
uint8_t FLASH.read(uint32_t address): Return a single byte from the FLASH buffer.
void FLASH.write(uint32_t address, uint8_t value): Write a single byte to the FLASH buffer.
bool FLASH.load(): Load content of FLASH into FLASH buffer.  Returns true if successful.
bool FLASH.save(): Save contents of FLASH buffer into FLASH.  Returns true if successful.
utnt32_t FLASH.length(): Static function that return the maximum available size of FLASH / buffer.

majenkotech, 2016-09-24:
You mean:

https://github.com/MajenkoLibraries/Flash

JacobChrist, 2016-09-24:
Make is so number 1.

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

JacobChrist, 2016-09-24:
Your library looks nice, it would be good if the function naming was more
inline with Wiring style names.  But I'll take it.  We should put this in
the core!

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

majenkotech, 2016-09-24:
By all means, help yourself. Feel free to rename the functions if you like too.

majenkotech, 2016-09-24:
By the way, it may need some work to make it go properly on the MZ - what with the caching and such that happens. Need to make sure everything goes through the right segments.