chipKIT32/chipKIT-core #275: Repeated compilations for Fubarino SD create error MSpanList_Insert

ricklon, 2016-10-19:
Arduino Version 1.6.11
Repeated compilations for Fubarino SD create error "MSpanList_Insert"
https://gist.github.com/ricklon/3c093a89eccaf945b2b8041dc4ec4ad3

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/rickanderson/Library/Arduino15/packages -hardware /Users/rickanderson/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/rickanderson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/rickanderson/Documents/Arduino/libraries -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10611 -build-path /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/rickanderson/projects/ricklon/autonomous/arduino/OTTOFullAutoDrive/OTTOFullAutoDrive.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/rickanderson/Library/Arduino15/packages -hardware /Users/rickanderson/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/rickanderson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/rickanderson/Documents/Arduino/libraries -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10611 -build-path /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/rickanderson/projects/ricklon/autonomous/arduino/OTTOFullAutoDrive/OTTOFullAutoDrive.ino
Using board 'fubarino_sd' from platform in folder: /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32
Using core 'pic32' from platform in folder: /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32
Detecting libraries used...
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "-I/Users/rickanderson/Documents/Arduino/libraries/Sparkfun_MPU-9250/src" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "-I/Users/rickanderson/Documents/Arduino/libraries/Sparkfun_MPU-9250/src" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SPI" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SPI" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/Wire" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "-I/Users/rickanderson/Documents/Arduino/libraries/Sparkfun_MPU-9250/src" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build16074617a2b163c1c137bf4bd1209a5c.tmp/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
failed MSpanList_Insert 0x31f728 0x2c9282bb24a3 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2cadd0, 0x31f728)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2cad60, 0x1, 0x10000000016, 0xf989)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2d2f58, 0x2d62f8)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x30c000, 0xc200000016, 0xc208012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc20826ae00 sp=0xc20826adf8
runtime.mallocgc(0x1a0, 0x0, 0xc200000003, 0xc208058e80)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc20826aeb0 sp=0xc20826ae00
runtime.rawstring(0x197, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:195 +0x93 fp=0xc20826aee0 sp=0xc20826aeb0
runtime.slicebytetostring(0xc208043d40, 0x197, 0x197, 0x0, 0x0)
	/usr/local/go/src/runtime/string.go:69 +0x4b fp=0xc20826af70 sp=0xc20826aee0
[... cut ...]
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:220 +0x49 fp=0xc20826bce8 sp=0xc20826bcc0
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:316 +0x171b fp=0xc20826bf98 sp=0xc20826bce8
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc20826bfe0 sp=0xc20826bf98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc20826bfe8 sp=0xc20826bfe0
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Fubarino SD (1.5).
```

ricklon, 2016-10-19:
Version 1.6.12 of Arduino gives this error:

```

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/rickanderson/Library/Arduino15/packages -hardware /Users/rickanderson/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/rickanderson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/rickanderson/Documents/Arduino/libraries -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10612 -build-path /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/rickanderson/projects/ricklon/autonomous/arduino/OTTOFullAutoDrive/OTTOFullAutoDrive.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/rickanderson/Library/Arduino15/packages -hardware /Users/rickanderson/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/rickanderson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/rickanderson/Documents/Arduino/libraries -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10612 -build-path /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Users/rickanderson/projects/ricklon/autonomous/arduino/OTTOFullAutoDrive/OTTOFullAutoDrive.ino
Using board 'fubarino_sd' from platform in folder: /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32
Using core 'pic32' from platform in folder: /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32
Detecting libraries used...
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10612 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10612 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10612 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/libraries/SoftPWMServo" "-I/Users/rickanderson/Documents/Arduino/libraries/Sparkfun_MPU-9250/src" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch/OTTOFullAutoDrive.ino.cpp" -o "/dev/null"
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10612 -D_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino  -D_USE_USB_FOR_SERIAL_ -I/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/arduino_build_316254/sketch "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "-I/Us [... cut ...]
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:124 +0xb90
arduino.cc/builder.RunBuilder(0xc420077500, 0xc42003fda8, 0x244c58)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:220 +0x35
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:316 +0x912
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Fubarino SD (1.5).
```

ricklon, 2016-10-19:
This get's worse in Arduino 1.6.12 and the nightly build.
Current fix is close and open Arduino application.

EmbeddedMan, 2016-10-19:
Any chance this is already listed somewhere on the Arduino Builder issue
list? There are a bunch of open issues there, but I can't quite tell if any
of them relate.

*Brian