chipKIT32/chipKIT-core #279: analogReference() does not work for WiFire or OpenScope boards

JacobChrist, 2016-10-24:
Notes from topic posted on forum:

Some magic tricks had to be performed when Digilent make the chipKIT WiFire due to building the boards with different revisions of uC Silicone. Because of this the ADC code is not quite up to snuff at this time. Today we had to use and external reference and we ended up adding the following code to setup() to get this to work. This was tested on both a Rev B and Rev C WiFire boards. This code will use +VREF and AVss as your referances. If you need a different configuration refer the the ADC datasheet for the MZ parts to determine how these bits need to be set.

#if defined(__32MZ2048ECG100__)
AD1CON3bits.VREFSEL = 0b001;
// no need for EF code on an EF processor, just use the compiler supplied headers
#elif defined(__32MZ2048EFG100__)
ADCCON3bits.VREFSEL = 0b001;
#endif

Jacob

EmbeddedMan, 2016-11-28:
Jacob, do your recent pull requests satisfy this issue?

JacobChrist, 2016-11-28:
No, I found this issue when doing the latest changes.  This is also not
fixed for the MZ implementation I did.  If I get a chance I can fix this.
We have it working on the project we are working on but we implemented it
outside of the core.

Jacob