chipKIT32/chipKIT-core #280: Linux64 binaries are compiled for Linux32

jonrei, 2016-10-24:
Hi,

I'm using the linux64 binaries available [here](http://chipkit.net/wiki/index.php?title=ChipKIT_core), and they are not really for linux64.
Checking a random binary:

    file chipkit-core64/pic32/compiler/pic32-tools/bin/pic32-cpp
 
results in

     chipkit-core64/pic32/compiler/pic32-tools/bin/pic32-cpp: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.0.0, stripped

The only file that seems to be compiled for 64-bit is `chipkit-core-32/pic32/tools/bin/pic32prog`,  which with `file` gives the output: 

    chipkit-core64/pic32/tools/bin/pic32prog: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=a5b9fd9167c0d1ce2f6830fbc220f28a395898ca, not stripped

Trying to run any of these binaries on Ubuntu 16.10 results in the cryptic message

    bash: ./pic32-cpp: No such file or directory

This can be resolved by adding 32-bit support by following the instructions [here](http://askubuntu.com/questions/454253/how-to-run-32-bit-app-in-ubuntu-64-bit), but if they are advertised as 64-bit, it would be nice if they actually were.

majenkotech, 2016-10-24:
The compiler is only available in 32-bit. I have campaigned for a 64-bit version but "it is too much work" apparently. Compiling for 64-bit would, according to Microchip, break things.

So we have to make do with the 32-bit version, which means you have to install 32-bit support on 64-bit Linux. This is something you should probably do anyway.

I can't recall all the packages you need, but you can start off by installing `libc6-i386` and take it from there. That should allow it to execute the binary, at which point it will then tell you if it needs any other 32-bit libraries when you run it.

Hopefully that should be all you need - looking at the output of `ldd pic32-g++` you see:

    $ ldd pic32-gcc
           linux-gate.so.1 (0xf76f2000)
            libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xf7508000)
            /lib/ld-linux.so.2 (0xf76f5000)

The only 32-bit library there is libc.so.6, so that should be all it wants.

jonrei, 2016-10-24:
I see, maybe there should be a note somewhere in the documentation though? It took me some time to figure out what was going on.

majenkotech, 2016-10-24:
Already done a few minutes ago :)

* http://chipkit.net/wiki/index.php?title=ChipKIT_core#64-Bit_Linux

closed 2016-10-24 by jonrei