chipKIT32/chipKIT-core #281: chipKIT Windows Stk500 Driver

JacobChrist, 2016-10-24:
The chipKIT Windows installer is built using .NET 3.5.  Windows 8+ computers do not have this installed by default (but do have 1.0 ,20 and 4.0).  It would be nice to rebuild using either an older or newer version of .NET.

majenkotech, 2016-10-24:
Alternatively re-write the whole thing as a Nullsoft installer script.

* http://nsis.sourceforge.net/Main_Page

JacobChrist, 2016-10-24:
Schmalz created the last exe, and its signed.  If its easy for him to update then that would be nice.  If not this maybe an alternative.  Also, to get around the issue we just install from the inf, however the chipKIT wiki says to use the installer which doesn't work on Windows 8 without installing .NET 3.5.

JacobChrist, 2016-10-24:
I'm in no need of this, just thought its an issue that might affect others.

majenkotech, 2016-10-24:
Since I have some minor experience with NSIS through using it for UECIDE maybe I could take a look at revamping the installer... From what I read though installing drivers is different from Vista onwards and NSIS doesn't directly support it, so calling an external (standard windows) binary to do the job is needed. Not sure how to make that decision in NSIS though...

EmbeddedMan, 2016-10-25:
Jacob,

Unfortunately, we get that installer straight from Microchip. I don't know
much about how to recompile it with different dependencies. Maybe if
somebody has more Windows .Net programming experience this would be a
trivial thing to change.

It would be great if we could do it as a Nullsoft script - that's how we do
the EggBot installer, and it works great. (Again, I didn't write it, but
the guy who did isn't a Windows programmer either.)

*Brian

EmbeddedMan, 2016-10-25:
Yeah, Microchip has a nice word document that explains how to take the .inf
file, modify it for your own uses, then wrap it up with their installer exe
and sign the whole thing with your own cert so that it won't trigger
warnings on Windows 8/10. So that's all I did - I bought the cert (which I
just renewed) and followed their instruction. I didn't have anything to do
with how their installer is compiled though.

JacobChrist, 2016-10-25:
If the code is available I can rebuild for 4.0.

Jacob

majenkotech, 2016-10-28:
So Windows has a program called "dpinst.exe" that installs drivers for you. Someone else has used that, along with nsis, to make an installer for their driver (MIT licensed script :) ).

I have taken that script and associated dpinst files and made a pair of installers for the chipKIT drivers (a 32-bit and a 64-bit installer). 

I suggest someone with admin privs makes a new "chipKIT Drivers" repo on here that we can work in. Then I will post this test version there for us to play with.

EmbeddedMan, 2016-10-28:
You mean something like this? https://github.com/chipKIT32/chipKIT-drivers

*Brian

majenkotech, 2016-10-28:
Yes, something exactly like that.

majenkotech, 2016-10-28:
Ok PR on the other repo made. One thing that may now want changing with the script is to make it get the driver files from the existing structure rather than having its own internal copy.