chipKIT32/chipKIT-core #292: Create PONTECH NoFire variant

JacobChrist, 2016-11-14:
This variant will be a prototype/template for a board that would use the same processor on a Digilent WiFire Rev C board but without the special ADC tricks created to support multiple chips on a single board.  This will allow for testing non board specific MZ EFG analogRead()'s on a WiFire board.  It will also serve as a working prototype for anyone wishing to create a variant based on the PIC32MZ2048EFG100 chip.

EmbeddedMan, 2016-11-20:
Love this idea Jacob. Excellent.

JacobChrist, 2016-11-20:
I am not in love with the name, but I didn't want to confuse WiFire users.
It works great, we have been using this def with WiFire boards.

EmbeddedMan, 2016-11-28:
This is now merged in.

closed 2016-11-28 by EmbeddedMan