chipKIT32/chipKIT-core #300: EEPROM: getBuffer() writes 0xFF values back into EEPROM table when page is erased

JacobChrist, 2016-11-17:
In Deeprom.c in EEPROM library utils.

It looks like this feature was in the code at one time then removed.  Uncomment code that prevents this so that final function looks like this:

``` c++
void getBuffer(uint8_t * buffer)
{
	eeSeg tempSeg;
	uint8_t tempData;

 	int i;

	//Cycle through buffer
	for(i=0; i < max_address; i++)
	{
		//If data in buffer does not equal 0xFF write to flash
		if(buffer[i] != 0xFF) {
			tempData = buffer[i];
			tempSeg = pack(i, tempData);
			writeFlashWord((void*)&eedata_addr[0][i],tempSeg.data);
		}
	}
}
```

EmbeddedMan, 2016-11-28:
Jacob,

Are you suggesting that the code get updated as your example shows? Or are you just pointing out something? I'm not quite sure what action needs to be taken to close this issue.

JacobChrist, 2016-11-28:
I am working on fixing this, not sure if it is even an issue. The code
changes I did before only prevent writing FF with an EEPROM.write().  This
prevents writing FF's after the table has been erased.