chipKIT32/chipKIT-core #301: pwm frequency won't change?

tanthua, 2016-11-18:
Hi. I'm using chipkit uc32. 
I picked this code up from another issue that was resolved here, and it was compiled and uploaded successfully. However, when I measured the frequency of pin 3 or pin 5, they remained the same around 498 Hz. I would like to change the frequency up to 12 kHz but never got it working. 

Also, I used the Arduino IDE with chipkit library. When I uploaded the sketch from "StandardFirmataForChipkit", the frequency of pwm dropped from 498Hz to 398Hz constantly. 

Can you give me explanations for those incidents and also provide me a complete example how to change the frequency of pin 3 or 5 of the chipkit on Arduino IDE?
```
void setup() {
  T2CONbits.ON = 0;         // Turn the timer off
  T2CONbits.TCKPS = 0b011;        // Set prescaler
  TMR2 = 0;                 // Clear the counter
  PR2 = 4096;               // Set the period
  T2CONbits.ON = 1;         // Turn the timer on

  T3CONbits.ON = 0;         // Turn the timer off
  T3CONbits.TCKPS = 0b011;        // Set prescaler
  TMR3 = 0;                 // Clear the counter
  PR3 = 4096;               // Set the period
  T3CONbits.ON = 1;         // Turn the timer on
}

void loop() {
  analogWrite(3, 127);
  analogWrite(5, 127);
}
```

majenkotech, 2016-11-18:
When you call analogWrite() it sets the frequency overwriting the settings you have just made. The latest version of the core has a function to set the frequency properly - #253 

    analogWriteFrequency(12000);

tanthua, 2016-11-18:
Thanks so much for the answer. 
Would it change the frequencies of all the OC1-5 or just one specific pin?

majenkotech, 2016-11-18:
All pins together - they all run off timer 2.

tanthua, 2016-11-19:
I got the test you provided in another post working. But when I put them in setup() in StandardFirmataChipKit, it was back to 500Hz again. So weird. I tried put them in both setup() and loop() but none worked.

majenkotech, 2016-11-19:
Firmata uses SoftPWMServo for the PWM to allow PWM on every pin on the board. That always runs at 500Hz regardless. 

    void servoWrite(byte pin, int value)
    {
      SoftPWMServoPWMWrite(PIN_TO_PWM(pin), value);
    }

tanthua, 2016-11-19:
Oh darn....It's very good to know. It's going to be tough :(. Thanks so much!!!

majenkotech, 2016-11-19:
Maybe try with your Linux VM?

tanthua, 2016-11-19:
Can you elaborate it a bit more? I'm not sure I get that.

majenkotech, 2016-11-20:
How did that get there?!

majenkotech, 2016-11-20:
In chipKIT we have two PWM systems - hardware, which can only go on the OCx pins, and software, driven by the Core Timer, which can be on any pin. Firmata, to give PWM on all pins, uses the software one. There's no way of changing the frequency for the software PWM.

tanthua, 2016-11-20:
I kind of get it now. So do you think it's possible if I use MPLAB to change the pwm frequency and then use firmata to do the rest of the work? Or the firmata will override the pwm frequency regardless of the changes?

majenkotech, 2016-11-20:
If you want a different frequency in Firmata you will have to change Firmata to use the hardware PWM (analogWrite()) instead of the software PWM (SoftPWMServoPWMWrite()) in the function I showed you above. Then you will be able to use analogWriteFrequency().

tanthua, 2016-11-20:
Got it! Will give it a try. Thanks so much! :D

tanthua, 2016-11-24:
So when I do analogChangeFrequency(val), how would the resolution change along without touching analogChangeResolution(val)? Would there be any conflict cases?