chipKIT32/chipKIT-core #306: Strange typedef in Deeprom.h?

JacobChrist, 2016-11-22:
Does anyone else find this typedef in Deeprom.h to be strange

``` c++
typedef uint32_t BOOL;
```

Its especially strange to me be because stdint.h is included in this file before this typedef.

majenkotech, 2016-11-22:
It's a typical Microchipism

JacobChrist, 2016-11-22:
Do you know if it serves any purpose other than upsetting my obsessive compulsive disorder to remove unnecessary code?

majenkotech, 2016-11-22:
It does the same to me. Much Microchip code us littered with such idiotic typedefs.

JacobChrist, 2016-11-22:
The whole thing blows up if you take it out... seems like a problem for another day.

JacobChrist, 2016-11-22:
I see what's going on here.. Its been too long since I touch c/c++.. I was confusing the order of the type being defined and what is already defined.  This is creating a BOOL type out of a uint32_t which comes from stdint.h.  I was thinking it was the other way around.  I am less concerned now but wonder if this is the best way to pick up the BOOL type?

majenkotech, 2016-11-22:
It's A way, certainly. Since there is no such thing really as a bool in c (there is in c++) crazy things get invented. I hate all-caps type names. They are loathsome.

EmbeddedMan, 2016-11-22:
Normally you'd just #include <stdbool.h> and get the 'bool' type, but for
some reason Microchip wants to use types that are in all caps and are named
non-standard things (like BYTE and WORD). I really hope that, in the
future, they're able to convert their code over to just <stdint.h> and
<stdbool.h> types, as that would make their code more compatible with the
style the rest of the embedded world now uses.

*Brian

EmbeddedMan, 2016-11-28:
Jacob, if there is more to discuss here with regards to the strange typedef you found, please continue the discussion. Otherwise I think you can close this issue.

JacobChrist, 2016-11-28:
This can probably be closed. But Matt brought up a good point. Should we
use BOOL or bool? In spite of what is probably the more elegant code we
should probably ask WWAD? What would Arduino Do for maximum compatibility.

majenkotech, 2018-02-16:
WWAD? They'd use `boolean` - which is what Java uses (and be incompatible with everyone else in the community). `bool` is what we should be using. It's the standard, and programs like `vim` highlight it properly, which they don't with `BOOL` and `boolean`.

EmbeddedMan, 2018-02-16:
Agreed. BOOL (as well as things like INT8) are Microchip special defines.
We should be sticking the standards in stdint.h and stdbool.h.

*Brian

majenkotech, 2018-02-16:
Maybe we should set out a standard coding style document that all commits to the core must conform to? That would annoy Digilent ;)

Things like:

* Only use stdint.h integers: `int8_t`, `int16_t`, `int32_t`, `uint8_t`, `uint16_t`, `uint32_t`
* Only use stdbool.h's `bool`
* Always use camelCase for variable and function names
* Always use UpperWordCase for class, type and struct names

... etc ...

EmbeddedMan, 2018-02-16:
Yes, these should be our standards, and we should do our best to enforce
them on PRs. I don't believe it going back and re-writing existing code
just for coding standards like this, but as we update things . . .

*Brian

JacobChrist, 2018-02-19:
These all sound fair.  But we should probably have a define for boolean for
compatibility if it doesn't already exist.

#define boolean  BOOL

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

majenkotech, 2018-02-19:
Boolean is in the core. Ardueenies use it all the time.

EmbeddedMan, 2018-03-24:
So, @majenkotech  and @JacobChrist , what's the best way to move forward with this issue so we can close it? 

Should I do a wholesale replacement of BOOL in all core and libraries with bool?

JacobChrist, 2018-03-24:
I think we should go for maximum compatibility with c standards without
sacrificing Arduino compatibility. For this instance I think boil would get
us closer to c standard libs w/o changing and Arduino compatibility, though
not necessarily improving Arduino compatibility.

Jacob