chipKIT32/chipKIT-core #309: Update serial objects to use DMA as per Keith's code

EmbeddedMan, 2016-12-01:
Hi Guy / Brian,
 
I few things I have discovered about Arduino and timing. If you load your sketch and open the Serial Monitor window you may not get the first few characters printed by your sketch. Here is what is going on. When Arduino programs it calls the Arduino builder which in turn calls PIC32Prog/AVRDude/DigilentPGM. During this time Arduino must close the serial port if the serial monitor is open so that PIC32Prog can open the COM port and program. When PIC32Prog and then the Arduino builder is done, the IDE must reopen the COM port so that serial data will be displayed. You will notice that is you have the serial monitor open while programming, the window does not close, but the history clears. This was unlike MPIDE, you had to close the serial monitor before you could program. 
 
With DigilentPGM, I program much faster, get out to the bootloader much faster, and start my sketch much faster; so much so that the ID
[DMASerialTest.zip](https://github.com/chipKIT32/chipKIT-core/files/625381/DMASerialTest.zip)
E has not had a chance to reopen the COM port after DigilentPGM/Builder has terminated; and the serial monitor misses the first few writes from my sketch. I had to put into my bootloader a 500ms delay before transferring control to the sketch after programming, this solves the problem. We could also put that in the init() routine.
 
This issue only occurs if you program. If you do not program the IDE keeps the COM port open and does not miss the first Serial.print. You can notice this by seeing that if you program, the serial monitor clears history, but if you just hit the reset button on your board, the serial monitor does not clear the history.
 
On another topic…
 
Because I want to offload as much processing to the peripherals on the OpenScope, I have made a DMA based Hardware Serial (DMASerial). This requires no interrupt routine as I just set up the DMA to trigger when a character comes in and transfer it to the ring buffer. I have the DMA set up to auto enable, so the ring buffer just keeps filling overwriting old data if no-one read it out before the buffer wraps. The code is much simpler and can run much –MUCH-- faster and requires no interrupt routine. I have attached some sample code that compiles for the WiFIRE (and I think will work on  your latest chipKIT core; works on 1.1.0).
 
If you want to use the DMASerial instead of HardwareSerial for your board, you can disable the instantiation of the Serial object simple by not defining SERx_BASE and SERx_VECTOR in your Boards_Defs.h. In the OpenScope I have replaced the Serial object, but kept the same object name, simply by renaming these #defines and then instantiating my own Serial Object. In the example I sent you I don’t have the extended features, but I will be adding the ability to transfer directly into a user supplied buffer for receive, or to DMA out of a user supplied buff for write. I need this because I am transferring 65K of sample data and I need very high transfer rates. All processing will happen in the DMA/UART peripherals with no CPU utilization. Currently I am running the OpenScope Serial at 1.25MBaud, and when complete I hope (if the FTDI can keep up) to sustain 2.5MBaud with no CPU utilization or breaks in transmissions.
 

 

 
Thanks
KeithV
Attachments area
[image]

squintz, 2017-06-15:
What's the latest news for DMA Serial? I have a need for speedy TCP to Serial pass-through and have a feeling DMA will be required. Is the code attached in DMASerialTest.zip the latest?

EmbeddedMan, 2017-06-15:
Unfortunately I haven't had any time to look at this. I'm assuming that the latest is in DMASerialTest.zip. Please give it a try, and if it works for you, feel free to send a pull request to chipKIT-core with the changes necessary  to get it into the core.