chipKIT32/chipKIT-core #317: analogRead() takes about 2.4ms on a MZ ECG silicone running  @ 200MHz.

JacobChrist, 2016-12-05:
This is really really slow considering that MZ EFG silicone takes about 4.2us @ 200MHz

EmbeddedMan, 2016-12-05:
Is this with your new non-blocking ADC read change?

majenkotech, 2016-12-05:
The whole EC ADC is so broken it has had to be fudged around - and that makes it really slow.

Not alot that can be done about that. Just read the comments in ECADC.c...

    // must throw out first 16 samples
    // keep the 17th. We can not access any ADC registers
    // however we can look at the DMA to see when things complete

and

            // we have noticed problems that the DMA channel is not always
            // triggered, so after a time out value, just try again.
            // fundamentally the problem is that AD1FLTR6bits.AFEN must
            // be reset for each oversample conversion, disable and reenabled.

            // we know a conversion is going to take 8000ns * 16, or 128 us
            // we don't want to check too often so DMA and ADC can work
            // but we don't want to wait too long and hold things up

The presence of a number of loops and `delayMicroseconds()` is a big cause of the delay.

JacobChrist, 2016-12-05:
No, see Matt's comments. Wasn't sure if anything could be done, but I wanted to capture the issue.

majenkotech, 2016-12-05:
Oh, and don't get your silicon and silicone mixed up... silicon's what chips are made of - silicone's what you put in breasts.

JacobChrist, 2016-12-05:
Apparently the MZ ECG is made of silicone!

majenkotech, 2016-12-05:
Ah, that explains why it went tits up then ;)

EmbeddedMan, 2016-12-05:
Oh, I knew this was coming. Being located in the LA area, I'm sure silicone
just comes up on Jacob's spellcheck much more often than silicon. :-)