chipKIT32/chipKIT-core #319: Unable to use "int main()" on chipkit boards

mskoczen, 2016-12-07:
In mpide a few versions ago you could bypass a lot of the core functions, (increase execution speed / reduce code size), by calling int main() instead of setup and loop. This still works for arduino boards but when I try a chipkit board I get an error.

[sketch_test.txt](https://github.com/chipKIT32/chipKIT-core/files/635425/sketch_test.txt)

c:\users\username\appdata\local\arduino15\packages\chipkit\tools\pic32-tools\1.42-pic32gcc\bin\bin\gcc\pic32mx\4.8.3\..\..\..\..\bin/pic32-ld.exe: MPIDE Version not specfied correctly

c:\users\username\appdata\local\arduino15\packages\chipkit\tools\pic32-tools\1.42-pic32gcc\bin\bin\gcc\pic32mx\4.8.3\..\..\..\..\bin/pic32-ld.exe: Link terminated due to previous error(s).

collect2.exe: error: ld returned 255 exit status

exit status 255
Error compiling.

majenkotech, 2016-12-07:
By providing your own `main()` you basically stop most of the rest of the core from being linked in. That includes a special variable that the linker script requires in order to build the image configuration table.

You will need to provide your own copy of it in your program in order for it to exist:

    extern const uint32_t _verMPIDE_Stub;
    const uint32_t __attribute__((section(".mpide_version"))) _verMPIDE_Stub = MPIDEVER; 

Note that `MPIDEVER` is a macro that should be provided by the IDE on the command line. You can, of course, replace it with any 32-bit value of your choice if you really want to.

mskoczen, 2016-12-07:
Adding that does make it compile but the result is still much larger than I expected
(compiled for Uno32) Sketch uses 1,564 bytes (1%) of program storage space. Maximum is 126,976 bytes.
(compiled for Arduino Uno) Sketch uses 138 bytes (0%) of program storage space. Maximum is 32,256 bytes.

EmbeddedMan, 2016-12-07:
I believe there are a number of parts of the chipKIT-core code that do not
drop out of the binary even if they are not explicitly called from the
sketch.

Arduino has made a number of decisions along the way to really minimize the
final executable size, while chipKIT-core has made different decisions to
maximize performance or prevent complex behavior that is hard for users to
understand, with the result that for the same sketch our output is much
larger than Arduino's. There are TONS of ways we could work to reduce out
output side, but at this point we have not put many resources into that
effort as it doesn't seem like a very big payoff compared to other things
we can work on.

*Brian

JacobChrist, 2016-12-07:
Weather this is fix or not now, I think we should leave this issue open until fixed since it highlights a difference between Arduino and chipKIT.