chipKIT32/chipKIT-core #326: UART hardware TX FIFO too small

Serveurperso, 2017-01-28:
I need to TX a large burst of (lidar) data inside a non blocking project.

The Arduino core code contains a nice little round robin data buffer where you can keep throwing data at it and the arduino code will read the data and process it in order.

HardwareSerial.cpp : #define SERIAL_BUFFER_SIZE 256
And it work.

On PIC32 core I discovered this inside HardwareSerial.cpp :

```
size_t HardwareSerial::write(uint8_t theChar)
{

	while ((uart->uxSta.reg & (1 << _UARTSTA_UTXBF)) != 0)	//check the UTXBF bit
  {
		//* wait for the transmitter buffer to have room
	}

	uart->uxTx.reg = theChar;
    return 1;
}
```
A bad blocking while loop freezing my code :(

There is a way to do like Arduino on ChipKit core ?

http://phoenix-mindgame.blogspot.fr/2014/09/pic32-uart-transmission-using-tx.html

majenkotech, 2017-01-28:
Not without a complete re-write of the UART code, no. Unlike the Arduino, the PIC32 has a hardware FIFO for the TX buffer which we use. Some chips have a 4-byte FIFO, some have an 8-byte FIFO. If you need more I suggest you use the TX interrupt to feed more data to the FIFO from your own buffer. On suitable chips you could even configure the DMA to do the transfer for you so it doesn't use any CPU time at all.

Serveurperso, 2017-01-28:
I think it's simpler for me to return to Arduino... 8 byte deep FIFO is like nothing and rewrite entire low level code for UART is better inside a MPLAB project

majenkotech, 2017-01-28:
It's not that hard to hook into the interrupt. The interrupt isn't actively being used, so you just define yourself an interrupt handler and attach it to the right interrupt and vector. Then you can feed data in to your heart's content in the background.

You can read all about interrupts in chipKIT here: http://chipkit.net/interrupts-made-easy-with-chipkit/

You'd need to check the board definitions for your board to find which physical UART is used for the pins you are connected to (it's in Board_Defs.h for the variant), but it should be very simple to do.

I'll see if I can come up with some example code for you.

majenkotech, 2017-01-28:
Hang on, actually both TX and RX use the same vector. That'll get tricky, since the RX ISR is already running on that vector... :(

Serveurperso, 2017-01-28:
while ((uart->uxSta.reg & (1 << _UARTSTA_UTXBF)) != 0) is an insult to PIC32 power because anywhere I need to write more than 8 byte there is a delay like years inside my code:(

I know the ChipKit interrupt very easy to use. I just need a software trick to never get while() on UART TX anymore !

majenkotech, 2017-01-28:
Here is a transmit-only implementation - writing your own ISR clobbers reception, but you can add that in your own ISR if you need it. I am, for simplicity, using my CircularBuffer.h library for this, but that's up to you. https://github.com/MajenkoLibraries/CircularBuffer

````C++
#include <CircularBuffer.h>

CircularBuffer<uint8_t> txBuffer(1024);

volatile bool txRunning = false;

void __USER_ISR txIsr() {
    if (getIntFlag(_UART1_TX_IRQ)) {
        if (txBuffer.available()) {
            U1TXREG = txBuffer.read();
            txRunning = true;
        } else {
            txRunning = false;
        }
        clearIntFlag(_UART1_TX_IRQ);
    }
    if (getIntFlag(_UART1_RX_IRQ)) {
        clearIntFlag(_UART1_RX_IRQ);
    }
}

void setup() {
    Serial.begin(115200);
    Serial.println("Haha");
    setIntVector(_UART_1_VECTOR, txIsr);
    setIntPriority(_UART1_TX_IRQ, 4, 0);
    setIntPriority(_UART1_RX_IRQ, 4, 0);
    setIntEnable(_UART1_TX_IRQ);
    setIntEnable(_UART1_RX_IRQ);
}

void loop() {
    delay(1000);
    uint32_t start = micros();
    print("Buffered writing done through your own ISR. There is no reception here though, but you can add that yourself to the ISR.\r\n");
    uint32_t end = micros();
    uint32_t len = end - start;
    char temp[30];
    sprintf(temp, "Time taken: %dus\r\n", len);
    print(temp);
}

void print(const char *txt) {
    for (int i = 0; i < strlen(txt); i++) {
        txBuffer.write(txt[i]);
        // Needed to kickstart the transmission
        if (txRunning == false) {
            U1TXREG = txBuffer.read();
            txRunning = true;
        }
    }
}
````
This is set up for an Uno32 using the FT232 UART interface (Serial) - change to the right UART registers etc for your chosen interface.

Serveurperso, 2017-01-28:
Look promising, Can I use it for all 4 UART same time and what about RX ?

majenkotech, 2017-01-28:
Yes, you can use all 4 UARTs at once - you need a separate ISR for each, of course.

When a character is received it will trigger the same ISR, and you can receive the character from the buffer. In the ISR at the moment is:
````C++
    if (getIntFlag(_UART1_RX_IRQ)) {
        clearIntFlag(_UART1_RX_IRQ);
    }
````
Just add your reception code in there to read from U1RXREG (or whatever UART you are using).

I am wondering if maybe I should create my own UART library that is separate from the internal UART system. That way it's easy to add all sorts of funky features to it without disrupting the core...

Serveurperso, 2017-01-28:
Thanks a lot now I work to integrate this inside my code...

Why you can't upgrade the core to do buffered TX ? This seems essential because AVR Arduino do it !

Serveurperso, 2017-01-28:
I don't need large buffer for RX, can I use this for TX only and use Serial.available/read() from framework ?

majenkotech, 2017-01-28:
It could be done, and it may be done. No one else has needed it before now though. There's many other things I'd like to be able to do, though, such as DMA transfers on chips that support it, which I don't think would really sit well in the core (non-standard functionality).

majenkotech, 2017-01-28:
No, Serial.read() will always return -1 since it isn't getting the data fed to it any more.

majenkotech, 2017-01-28:
I have made a new library, "Pipe", which works like CircularBuffer.h but isn't templated. It inherits from Stream, so all the .print() functions etc work on it. It also has blocking writes when the buffer is full if you want them (second parameter `true` to the constructor):

* https://github.com/MajenkoLibraries/Pipe

An updated test sketch to go with it that does reception as well:
````C++
#include <Pipe.h>

Pipe txBuffer(1024, true);
Pipe rxBuffer(64);

volatile bool txRunning = false;

void __USER_ISR txIsr() {
    if (getIntFlag(_UART1_TX_IRQ)) {
        if (txBuffer.available()) {
            U1TXREG = txBuffer.read();
            txRunning = true;
        } else {
            txRunning = false;
        }
        clearIntFlag(_UART1_TX_IRQ);
    }
    if (getIntFlag(_UART1_RX_IRQ)) {
        rxBuffer.write(U1RXREG);
        clearIntFlag(_UART1_RX_IRQ);
    }
}

void setup() {
    Serial.begin(115200);
    setIntVector(_UART_1_VECTOR, txIsr);
    setIntPriority(_UART1_TX_IRQ, 4, 0);
    setIntPriority(_UART1_RX_IRQ, 4, 0);
    setIntEnable(_UART1_TX_IRQ);
    setIntEnable(_UART1_RX_IRQ);
}

void loop() {
    delay(1000);
    uint32_t start = micros();
    txBuffer.println("Buffered writing done through your own ISR. There is no reception here though, but you can add that yourself to the ISR.");
    if (txRunning == false) {
        U1TXREG = txBuffer.read();
        txRunning = true;
    }
    uint32_t end = micros();
    uint32_t len = end - start;
    txBuffer.print("Time taken: ");
    txBuffer.print(len);
    txBuffer.println("us");

    while (rxBuffer.available()) {
        txBuffer.write(rxBuffer.read());
    }
}
````

majenkotech, 2017-01-28:
Ok... new update to Pipe that has an `onWrite()` callback function. It also has an updated version of the above code which uses it as an example sketch.

Serveurperso, 2017-01-28:
Sorry I don't understand the "it can lock up with blocking writes if you write too much before kicking the transmission off" (I'm french with bad english)
I test it

majenkotech, 2017-01-28:
The example with the Pipe library doesn't suffer from that problem. The
issue was if you tried writing more data to the Pipe than would fit in it
before the Pipe had started to be emptied by the interrupt.

Serveurperso, 2017-01-28:
Yes I edited because I understood when I read the code:)
My first implementation of your lib inside my code don't work but I think there is a mistake with UART2 (Serial1) register naming

Serveurperso, 2017-01-28:
_UART2* / U2?XREG is for Serial1 and not Serial2

Serveurperso, 2017-01-28:
_UART_1B_VECTOR freeze my system !

majenkotech, 2017-01-28:
Which board are you using?

Serveurperso, 2017-01-28:
UBW32 with pic32mx795f512l and Max32 UART bootloader

majenkotech, 2017-01-28:
Hmm, I don't have one of those to test with...

Serveurperso, 2017-01-28:
Like a Max32... 4 UART U1A U1B U3A U3B

majenkotech, 2017-01-28:
I see the board definition only has 2 uarts defined - U1 and U2. Unless you are using the max32 definition with it?

Serveurperso, 2017-01-28:
I use Max32 definition with it !

Serveurperso, 2017-01-28:
This is why I use Max32 for all. This is the only bootloader with UART programming (for wireless programming) and 4 UART defined

majenkotech, 2017-01-28:
U1, U4, U2, U5, are serial, serial1, serial2 and serial3.

Serveurperso, 2017-01-28:
lol !!! I test. I think you solved the problem

Serveurperso, 2017-01-28:
UART4/U4 freeze the system also:(

majenkotech, 2017-01-28:
You are starting the serial port with Serial1.Begin, yes?

Serveurperso, 2017-01-28:
Yes
(cut past for my test code)
```
#include <SimpleTimer.h>
#include <Pipe.h>

#define DEBUG Serial
#define MODEM Serial1
#define SERVOS Serial2
#define LIDAR Serial3

void __USER_ISR txIsr() {
    if (getIntFlag(_UART4_TX_IRQ)) {
        if (txBuffer.available()) {
            U4TXREG = txBuffer.read();
            txRunning = true;
        } else {
            txRunning = false;
        }
        clearIntFlag(_UART4_TX_IRQ);
    }
    if (getIntFlag(_UART4_RX_IRQ)) {
        rxBuffer.write(U4RXREG);
        clearIntFlag(_UART4_RX_IRQ);
    }
}

void kick() {
    if (txRunning == false) {
        U4TXREG = txBuffer.read();
        txRunning = true;
    }    
}

void setup() {
 DEBUG.begin(115200);
 DEBUG.println("setup()");
 MODEM.begin(115200);
 txBuffer.onWrite(kick);
 setIntVector(_UART_4_VECTOR, txIsr);
 setIntPriority(_UART4_TX_IRQ, 4, 0);
 setIntPriority(_UART4_RX_IRQ, 4, 0);
 setIntEnable(_UART4_TX_IRQ);
 setIntEnable(_UART4_RX_IRQ);
 SERVOS.begin(1000000);
 LIDAR.begin(115200);
```

majenkotech, 2017-01-28:
I am not at my desk at the moment so I can't test. When I get back there later on I will dig out my max32 to have a play.

Serveurperso, 2017-01-28:
/* Serial port 1 uses UART4 (aka UART1B)
*/
#define	_SER1_BASE		_UART4_BASE_ADDRESS
#define	_SER1_IRQ		_UART4_ERR_IRQ
#define	_SER1_VECTOR	_UART_4_VECTOR
#define _SER1_IPL_ISR	IPL2SOFT
#define	_SER1_IPL		2
#define	_SER1_SPL		0

I don't understand why this freeze entire system

majenkotech, 2017-01-28:
Well, this is odd. UART4's TX interrupt looks like it's permanently set and will never clear...

majenkotech, 2017-01-28:
Oh yeah, I know what this is... The interrupt *keeps triggering while the TX FIFO is empty* - something that is different between the 4-byte and 8-byte deep FIFOs, which is why it worked on the Uno32 but not the Max32.

So you just need to start with the TX IRQ disabled, and turn it on in the txBuffer's "onWrite()" callback - and then turn it off in the ISR if there is nothing left to transmit.

Just working on some sample code...

majenkotech, 2017-01-28:
````C++
#include <Pipe.h>

Pipe txBuffer(1024, false);
Pipe rxBuffer(64);

volatile bool txRunning = false;

void __USER_ISR txIsr() {
    digitalWrite(13, HIGH);
    if (getIntFlag(_UART4_TX_IRQ)) {
        if (txBuffer.available()) {
            U4TXREG = txBuffer.read();    
            txRunning = true;
        } else {
            clearIntEnable(_UART4_TX_IRQ);
            txRunning = false;
        }
        clearIntFlag(_UART4_TX_IRQ);
    }
    
    if (getIntFlag(_UART4_RX_IRQ)) {
        rxBuffer.write(U4RXREG);
        clearIntFlag(_UART4_RX_IRQ);
    }

    if (getIntFlag(_UART4_ERR_IRQ)) {
        clearIntFlag(_UART4_ERR_IRQ);
    }
        
    digitalWrite(13, LOW);
}

void kick() {
    if (!txRunning) {
        txRunning = true;
        U4TXREG = txBuffer.read();    
        setIntEnable(_UART4_TX_IRQ);
    }    
}

void setup() {
    Serial.begin(115200);
    Serial1.begin(115200);
    Serial2.begin(115200);
    Serial3.begin(115200);
    pinMode(13, OUTPUT); digitalWrite(13, LOW);
    txBuffer.onWrite(kick);
    setIntVector(_UART_4_VECTOR, txIsr);
    setIntPriority(_UART_4_VECTOR, 2, 0);
    clearIntFlag(_UART4_TX_IRQ);
    clearIntFlag(_UART4_RX_IRQ);
    clearIntFlag(_UART4_ERR_IRQ);    
    setIntEnable(_UART4_RX_IRQ);
    clearIntEnable(_UART4_TX_IRQ);
}

void loop() {
    static uint32_t ts = millis();
    if (millis() - ts >= 1000) {
        ts = millis();
        Serial.println(ts);
        txBuffer.println(ts);
    }
    while (Serial.available()) {
        txBuffer.write(Serial.read());
    }
    while (rxBuffer.available()) {
        Serial.write(rxBuffer.read());
    }
}
````

Serveurperso, 2017-01-29:
Thanks a lot for your work, I test this now

Serveurperso, 2017-01-29:
There is no more freeze but I get a lot of incomprehensible data corruption

majenkotech, 2017-01-29:
Do you have blocking writes turned on for the TX pipe?

Serveurperso, 2017-01-29:
Pipe txBuffer(256, true);
Pipe rxBuffer(64);

Serveurperso, 2017-01-29:
There is a "protocol analyser" on my project, this look a timing problem. The RX side of my modem need a perfect timing, no "hole" in data flow else it transmit on radio side.

Serveurperso, 2017-01-29:
Also I just found the first byte at the end of all radio frames.

Serveurperso, 2017-01-29:
I transmit @ 20Hz and each (digital) frame begin with ascii "$R" (sync bytes).
I found the "$" at the end of each frame. And a 1 byte left or right rotation for all recovered data.

Serveurperso, 2017-01-29:
I receive @ 10Hz in place of 20Hz - because the decoder can sync one frame for two transmited.

Serveurperso, 2017-01-29:
The RX side of your sample work perfectly, the TX side have a 1 byte bug

majenkotech, 2017-01-29:
How big are the packets you are sending?

Serveurperso, 2017-01-29:
Tested with fixed 53 bytes.

Serveurperso, 2017-01-29:
The problem can be mine, I must restart all to test

Serveurperso, 2017-01-29:
No, this is not a problem here because it work when I return to the framework Serial1.write.

Serveurperso, 2017-01-29:
I get exactly the same problem with a smaller frame size:
I get the "$" at end, and 1/2 frame out of sync.

```
ModemRX | $RÂ..WN..H....àNÝH$ | 24 52 C2 00 00 57 4E 0A 09 48 80 00 80 00 E0 4E DD 48 24 | 036 082 194 000 000 087 078 010 009 072 128 000 128 000 224 078 221 072 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $R...X+..H....àOÞ^$ | 24 52 BF 00 00 58 2B 0A 04 48 80 00 80 00 E0 4F DE 5E 24 | 036 082 191 000 000 088 043 010 004 072 128 000 128 000 224 079 222 094 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wø..H....àOÞF$ | 24 52 BF 00 00 57 F8 0A 02 48 80 00 80 00 E0 4F DE 46 24 | 036 082 191 000 000 087 248 010 002 072 128 000 128 000 224 079 222 070 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $R...W...H....àLÞZ$ | 24 52 BF 00 00 57 B9 0A 03 48 80 00 80 00 E0 4C DE 5A 24 | 036 082 191 000 000 087 185 010 003 072 128 000 128 000 224 076 222 090 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $RÂ..W...H....àRÝN$ | 24 52 C2 00 00 57 88 0A 05 48 80 00 80 00 E0 52 DD 4E 24 | 036 082 194 000 000 087 136 010 005 072 128 000 128 000 224 082 221 078 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $RÂ..WW..H....àPÝK$ | 24 52 C2 00 00 57 57 0A 07 48 80 00 80 00 E0 50 DD 4B 24 | 036 082 194 000 000 087 087 010 007 072 128 000 128 000 224 080 221 075 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $RÂ..W0..H....àJÞH$ | 24 52 C2 00 00 57 30 0A 09 48 80 00 80 00 E0 4A DE 48 24 | 036 082 194 000 000 087 048 010 009 072 128 000 128 000 224 074 222 072 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wö..H....àTÞM$ | 24 52 BF 00 00 57 F6 0A 02 48 80 00 80 00 E0 54 DE 4D 24 | 036 082 191 000 000 087 246 010 002 072 128 000 128 000 224 084 222 077 036
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemTX | $S.ÿ.ÿ....ì.. | 24 53 01 FF 01 FF 03 80 80 80 EC 05 0A | 036 083 001 255 001 255 003 128 128 128 236 005 010
ModemRX | $R...WÞ..H....àMÞZ$ | 24 52 BF 00 00 57 DE 0A 03 48 80 00 80 00 E0 4D DE 5A 24 | 036 082 191 000 000 087 222 010 003 072 128 000 128 000 224 077 222 090 036
```

Serveurperso, 2017-01-29:
With framework serial, correct TDD communication is:

```
ModemRX | $R...Ws........àQÞN | 24 52 BC 00 00 57 73 0A 05 00 00 80 00 80 00 E0 51 DE 4E | 036 082 188 000 000 087 115 010 005 000 000 128 000 128 000 224 081 222 078
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wc........áQÞ[ | 24 52 BD 00 00 57 63 0A 06 00 00 80 00 80 00 E1 51 DE 5B | 036 082 189 000 000 087 099 010 006 000 000 128 000 128 000 225 081 222 091
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wã........àRÝR | 24 52 BC 00 00 57 E3 0A 04 00 00 80 00 80 00 E0 52 DD 52 | 036 082 188 000 000 087 227 010 004 000 000 128 000 128 000 224 082 221 082
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wn........álÝ. | 24 52 BD 00 00 57 6E 0A 06 00 00 80 00 80 00 E1 6C DD 8F | 036 082 189 000 000 087 110 010 006 000 000 128 000 128 000 225 108 221 143
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...X.........àlÞh | 24 52 BF 00 00 58 0E 0A 01 00 00 80 00 80 00 E0 6C DE 68 | 036 082 191 000 000 088 014 010 001 000 000 128 000 128 000 224 108 222 104
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $RÀ..W.........àqÜg | 24 52 C0 00 00 57 B5 0A 03 00 00 80 00 80 00 E0 71 DC 67 | 036 082 192 000 000 087 181 010 003 000 000 128 000 128 000 224 113 220 103
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wg........àeÝ. | 24 52 BF 00 00 57 67 0A 06 00 00 80 00 80 00 E0 65 DD 90 | 036 082 191 000 000 087 103 010 006 000 000 128 000 128 000 224 101 221 144
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $RÀ..W7........àKÞH | 24 52 C0 00 00 57 37 0A 09 00 00 80 00 80 00 E0 4B DE 48 | 036 082 192 000 000 087 055 010 009 000 000 128 000 128 000 224 075 222 072
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $RÀ..X.........àJÞM | 24 52 C0 00 00 58 09 0A 03 00 00 80 00 80 00 E0 4A DE 4D | 036 082 192 000 000 088 009 010 003 000 000 128 000 128 000 224 074 222 077
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wý........àRÞN | 24 52 BF 00 00 57 FD 0A 03 00 00 80 00 80 00 E0 52 DE 4E | 036 082 191 000 000 087 253 010 003 000 000 128 000 128 000 224 082 222 078
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...X.........àWÝ^ | 24 52 BF 00 00 58 04 0A 03 00 00 80 00 80 00 E0 57 DD 5E | 036 082 191 000 000 088 004 010 003 000 000 128 000 128 000 224 087 221 094
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...W+........àSÞX | 24 52 BD 00 00 57 2B 0A 09 00 00 80 00 80 00 E0 53 DE 58 | 036 082 189 000 000 087 043 010 009 000 000 128 000 128 000 224 083 222 088
ModemTX | $S.ø.ÿ....ì.. | 24 53 01 F8 01 FF 03 80 80 80 EC 05 0A | 036 083 001 248 001 255 003 128 128 128 236 005 010
ModemRX | $R...Wi........àQÞ_ | 24 52 BC 00 00 57 69 0A 06 00 00 80 00 80 00 E0 51 DE 5F | 036 082 188 000 000 087 105 010 006 000 000 128 000 128 000 224 081 222 095
```

majenkotech, 2017-01-29:
Any chance you could provide an MCVE for me to test with?

Serveurperso, 2017-01-29:
To do a good MCVE we need a RF modem like mine. This is a HM-TRP (used on 3DR Radio) with a custom firmware from me.

The firmware wait for UART data and transmit over the air if there is data inside the buffer and no more data coming for more than a millisecond

It's easy to make a mock-modem with another board like Arduino.

Serveurperso, 2017-01-29:
Why there is a hole from >1ms inside your TX ?

majenkotech, 2017-01-29:
A hole? how do you mean?

Serveurperso, 2017-01-29:
On the modem :

if(serial_len && timer2_tick() - serial_time > SERIAL_TIMEOUT_TICKS) {
 transmit...

Serveurperso, 2017-01-29:
On the PIC32 :
```
void writeModem() {
 uint8_t v8;
 uint16_t theta16;
 uint16_t odometriex16;
 uint16_t odometriey16;

 v8 = map(max(v, VBATMIN), VBATMIN, VBATMAX, 0, 255);
 theta16 = int(theta / (2.0 * PI) * 65536.0);
 odometriex16 = int(odometriex * 1000.0) + 0x8000;
 odometriey16 = int(odometriey * 1000.0) + 0x8000;

 MODEM.write("$R");
 MODEM.write(v8);
 MODEM.write(pwmMax);
 MODEM.write(erreurMax);

 for(uint8_t i = 0; i < NBMESURESTX; i++) {
  MODEM.write(angles_q6[i] >> 8);
  MODEM.write(angles_q6[i] & 0xFF);
  MODEM.write(distances_q2[i] >> 8);
  MODEM.write(distances_q2[i] & 0xFF);
 }

 MODEM.write(theta16 >> 8);
 MODEM.write(theta16 & 0xFF);
 MODEM.write(odometriex16 >> 8);
 MODEM.write(odometriex16 & 0xFF);
 MODEM.write(odometriey16 >> 8);
 MODEM.write(odometriey16 & 0xFF);
 MODEM.write(TXCHANNEL);
 MODEM.write(RXCHANNEL);
}
```

Serveurperso, 2017-01-29:
MODEM.write for the framework serial.
I replace 
MODEM.write("$R");
with
txBuffer.write('$');
txBuffer.write('R');
etc...

Serveurperso, 2017-01-29:
I think there is a 1 byte missing or lasting from previous TX, when it do a new TX.
You can't show it because I think you use Arduino terminal with hidden return invisible char.
We must test your code with binary data and RX with a hex editor.

majenkotech, 2017-01-29:
Nope, not seeing it at all. And no, I'm not using the Arduino IDE's shoddy excuse for a serial terminal. I'm using a logic analyser to interrogate the waveform (like Trump waterboarding an innocent Muslim).

[image]

There I am sending the same $R... packet (from your example output above) over and over again with a 2ms delay between them - using the example code I posted above.

It could be that your TX interrupt is being interrupted by another interrupt (UART4 has a lower "natural" priority than UARTs 1-3), so you could try increasing the priority from 2 (which all UARTs are on by default) to something higher (up to 7, which is maximum priority).

[167 more replies]