chipKIT32/chipKIT-core #327: Add a set of chipKIT variant defines

rmcolbert, 2017-02-08:
Arduino also defines board names for conditional compiling.
One example:
`#define *VARIANT_CHIPKIT_MAX32*
`I'm making a cross-variant compatible robotic arm firmware and things like
changing the analogWriteResolution is helpful but doesn't exist on many of
the AVR boards. For the DUE, it was simple:
```
#if defined(_VARIANT_ARDUINO_DUE_X_)
analogWriteResolution(12); // for Due
#endif
```
It would be nice to have for the ChipKIT boards.

JacobChrist, 2017-02-08:
These exist... I'm sure someone else will chime in before I find where this is documented, but if not I will see what I can dig up.

JacobChrist, 2017-02-08:
Actually this comment from Brian makes me think we do not now:

Robert - that sounds like a great idea. Can you file a separate issue for
adding a set of chipKIT variant defines that you can use in your sketches?
That way we can track is separately from the missing pins issue.

majenkotech, 2017-02-08:
You mean these?

```
cerebot32mx4.build.board=_BOARD_CEREBOT_32MX4_
cerebot32mx7.build.board=_BOARD_CEREBOT_32MX7_
cerebot_mx3ck.build.board=_BOARD_CEREBOT_MX3CK_
chipkit_mx3.build.board=_BOARD_CEREBOT_MX3CK_
cerebot_mx3ck_512.build.board=_BOARD_CEREBOT_MX3CK_512_
cerebot_mx4ck.build.board=_BOARD_CEREBOT_MX4CK_
chipkit_pro_mx4.build.board=_BOARD_CEREBOT_MX4CK_
cerebot_mx7ck.build.board=_BOARD_CEREBOT_MX7CK_
chipkit_pro_mx7.build.board=_BOARD_CEREBOT_MX7CK_
chipkit_Pi.build.board=_BOARD_CHIPKIT_PI_
chipkit_Pi_USB_Serial.build.board=_BOARD_CHIPKIT_PI_
cmod.build.board=_BOARD_CMOD_
CUI32stem.build.board=_BOARD_CUI32_
ubw32_mx460.build.board=_BOARD_UBW32_MX460_
ubw32_mx795.build.board=_BOARD_UBW32_MX795_
cui32.build.board=_BOARD_CUI32_
chipkit_DP32.build.board=_BOARD_DP32_
fubarino_mini_dev.build.board=_BOARD_FUBARINO_MINI_
fubarino_mini.build.board=_BOARD_FUBARINO_MINI_
#fubarino_mini_dbg.build.board=_BOARD_FUBARINO_MINI_
fubarino_sd_seeed.build.board=_BOARD_FUBARINO_SD_
fubarino_sd.build.board=_BOARD_FUBARINO_SD_
Fubarino_SDZ.build.board=_BOARD_FUBARINO_SDZ_
mega_pic32.build.board=_BOARD_MEGA_
mega_usb_pic32.build.board=_BOARD_MEGA_
Olimex_Pinguino32.build.board=_BOARD_OLIMEX_PIC32_PINGUINO_
picadillo_35t.build.board=_BOARD_PICADILLO_35T_
pontech_NoFire.build.board=_BOARD_PONTECH_NOFIRE_
quick240_usb_pic32.build.board=_BOARD_PONTECH_QUICK240_USB_
usbono_pic32.build.board=_BOARD_PONTECH_UAV100_
chipkit_uc32.build.board=_BOARD_UC32_
uc32_pmod.build.board=_BOARD_UC32_
uno_pic32.build.board=_BOARD_UNO_
uno_pmod.build.board=_BOARD_UNO_
chipkit_WF32.build.board=_BOARD_WF32_
chipkit_WiFire.build.board=_BOARD_WIFIRE_
chipkit_WiFire_AB.build.board=_BOARD_WIFIRE_
chipkit_WiFire_80MHz.build.board=_BOARD_WIFIRE_80MHZ_
OpenScope.build.board=_BOARD_OPENSCOPE_
openbci.build.board=_BOARD_DP32_
lenny.build.board=_BOARD_LENNY_
clicker2.build.board=_BOARD_CLICKER2_PIC32MX_
```

EmbeddedMan, 2017-02-08:
Ha! All that means is that I don't remember if we do or not. :-) The first
part of working on that issue will be to see if we already do have
something in place (I believe we do actually), then documenting it
somewhere nice (like the wiki).

EmbeddedMan, 2017-02-08:
Are those all available to the sketch?

majenkotech, 2017-02-08:
Yep. 

> recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}"  {compiler.cpp.flags} -mprocessor={build.mcu} -DF_CPU={build.f_cpu}  -DARDUINO={runtime.ide.version} **-D{build.board}** {compiler.define} "{compiler.cpp.extra_flags}" {build.extra_flags} -I{build.path}/sketch {includes} "{source_file}" -o "{object_file}"

JacobChrist, 2017-02-08:
This is exactly what I was thinking of... I could find the variants file fast enough...

JacobChrist, 2017-02-08:
There is also the generic ones like:

#if defined(__PIC32_PPS__)
#if defined(__PIC32MZXX__)

ect...

majenkotech, 2017-02-08:
and of course 

```
#ifdef __PIC32__
#ifdef __PIC32MX__
#ifdef __PIC32MZ__
```
and many others.

EmbeddedMan, 2017-02-08:
These should all go on the wiki

majenkotech, 2017-02-08:
The board specific ones should be mentioned in each board page, certainly. Maybe add them to the right-hand box.

JacobChrist, 2017-02-08:
Some were on the old wiki (which means they are somewhere in wordpress).  

I found these but I think they are out of date:

http://chipkit.net/programming-hints/
http://chipkit.net/tips-porting-arduino-libraries/
http://chipkit.net/header-files/

majenkotech, 2017-02-08:
I have added it the Lenny page and created a page telling you what it is.

* http://chipkit.net/wiki/index.php?title=ChipKIT_Lenny

Look in the right-hand panel.

rmcolbert, 2017-02-08:
I saw `_BOARD_MEGA_` in the variants file but thought that I had to be misunderstanding it because it doesn't match the board name ChipKIT Max32 and seemed too generic. I'll probably try to use the chip type generic instead.

majenkotech, 2017-02-08:
For some strange reason when the UNO32 and MAX32 were first created they chose the defines that are the same as the Arduino equivalents. Bit silly, really...

majenkotech, 2017-02-08:
Of course you can combine flags:

```c
#if defined(__PIC32__) && defined(_BOARD_MEGA_)
// This will be the MAX32 and not the Arduino Mega2560.
#endif

JacobChrist, 2017-02-09:
Digilent or Microchip one also claimed 100% Arduino compatible (don'r remember who).  Which was even sillier since the boards and there accomplishment was so great regardless of the level of compatibility.

majenkotech, 2017-02-09:
Fortunately the whole market has moved away from 100% Arduino Compatibility now - with so many ARM chips being used in Arduino-like products the only boards now that are 100% Arduino compatible are Arduino AVR boards. Everything else is just a close approximation, and people are getting used to that idea. 

So while we should all strive to get as close as possible with the core functions the overall system doesn't have to be 100% exact any more. 

The one place we are currently failing though is in networking. Yes, Keith et al did a fantastic job on making a very efficient and robust networking stack for chipKIT, but it's nigh on impossible for a novice user to use, and can never work with any other Arduino libraries that expect the normal Client / Server interface - so things like PubSubClient, which is getting very popular now, won't work. Such a shame. I really wish I had time to write my own modular implementation from the ground up in the style of DisplayCore. I did write my own IP stack once upon a time from scratch to work with the ENC25J60 chips. Maybe I could dig that old stuff out and build on it...

rmcolbert, 2017-02-09:
Thank you all for the information. Do you want to keep this issue open as a reminder to add information to the board wiki pages or close it out? Either works for me.

EmbeddedMan, 2017-02-09:
The issue should be kept open, as this information needs to be provided for
each board on it's own Wiki page, and we need to have some sort of central
place that can direct people to the various other places that the actual
defines are listed.

squintz, 2017-04-09:
I have added the information to each boards wiki page. Please review them to make sure I got the right. I considered making a central location to make a list as well but that would mean two places to maintain the same data.

EmbeddedMan, 2017-04-09:
Fantastic! I think having them in the Wiki board pages is great - they look good. 

What about all of the non-board specific defines though? Where should they go?

JacobChrist, 2017-04-09:
Yes, they are looking amazing!

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com