chipKIT32/chipKIT-core #335: Servo library appears to not work with Wi-Fire

squintz, 2017-03-16:
Although there are no warnings or errors when I compile there seems to be something wrong with the Servo library when used with the Wi-Fire board. The SoftPWMServo library functions normally while the Servo library seems to not be generating the PWM signal. Initially I was attempting to use the hardware PWM on Pin 3. When that did not work I walked my signal line around to all the other pins to see if for some reason the pins were not mapped correctly but I did not find a PWM signal on any pin. When I ran the same code on a Uno32 board the Servo turned as expected and therefore I've ruled out a bad Servo. I also verified power to the Servo with a volt meter.

From Marshall W. via E-mail:

"To confirm, softPWMServo worked on pin 3 but the OC Servo didn't work?

If that is true then I have a couple ideas. The PPS may not be set correctly to map an OC to RPD0. The other possible situation would be if the RTCC or interrupt modules are taking priority over the RPD0.

The best way to check this would be to look into the board files and recheck that the attributes for that pin are correctly labeled with the OC/PWM flag (I don't remember what it is called but it is obvious).  Then double-check whether the RTCC and interrupt modules take priority or are initialized in an active state.

I suspect one of those two are the issue. I haven't had to dig into the MZ board files but that is what I would suspect to run down on an MX. Sorry if this is too obvious. I was hoping to help.
"

squintz, 2017-10-21:
Bump!

Has anyone looked into making a library that support Hardware PWM instead of using timers>. I'm building a robot with 5 motors and would love to have something that doesn't eat all the cycles just to control 5 separate PWMs.

EmbeddedMan, 2017-10-21:
Hi! Are these normal RC servos you're talking about?

If so, the SoftPWMServo library that's included with chipKIT should fit
your bill. It does not use any timer peripherals, it is completely software
based, and it can generate 84 simultaneous RC servo signals while only
using about 10% of an 80MHz PIC32MX core. (So, on the faster MZ core, it
will be far less.) And if you only need 5 servo outputs, you probably won't
even be able to measure the CPU hit it will be so small.

That said, you are right in that it would be very nice if we got the
standard Servo library working on MZ cores too.

*Brian

squintz, 2017-10-21:
You know what! I must have had some wires crossed in my brain today. I'm
using DC motors with an H-Bridge and so now that I'm looking at
analogWrite() I think I"m going to be ok.

David

bamberjo, 2018-08-01:
Bump again!

Hello,

I am attempting to use a the rev C. chipkit WiFire as the microprocessor for a closed loop control system for an active damping system using 6 brushless motors with ESC's that I am trying to interface with using the chipkit servo library. I have been experiencing the same problem described above that although there are no compiling errors using the servo library, when I probe the outputs of the chipkit I am not able to find a signal on any of the PWM pins. However, when I switch to softPWMservo I am able to successfully create the required signal.

As latency is an important consideration in the control system, we were hoping to use the hardware PWM. I was wondering anyone has any progress on how to successfully use the hardware for creating the RC PWM signal. If not, does anyone have any data/thoughts on how much latency using the softPWMServo library for independent control of 6 ESC's might introduce and if it will be a big problem given the 200MHz speed of the WiFire.

Thanks,

Josh

EmbeddedMan, 2018-08-01:
Josh,

Have you measured the latency of SoftPWMServo? I'm curious, because
especially on the WiFire, it should be very, very low. I expect it to be
the same latency as with hardware PWM - in other words, the time from when
you make the function call to update the output to a new value until when
the output actually changes to the new value will be the same between the
two. At least that's my expectation. I have not actually done that
experiment, but I did write the SoftPWMServo library, so I have done a
number of experiments and I understand the underlying architecture.

The only disadvantage of the SoftPWMServo library is jitter, and that may
or may not be even measureable on your system once the signals get through
the ESC. (Some ESCs quantize the incoming signal.)

*Brian

bamberjo, 2018-08-01:
Thanks for the prompt reply,

I have not measured the latency of SoftPWMServo. I am definitely a beginner when it comes to the nitty gritty of microcontrollers and had thought that there was a large difference between the speed of hardware PWM and software PWM, but if the latency should be fairly negligible then we should be able to use SoftPWMServo instead.

In terms of the jitter, I am not sure of the specifics of our ESC, but I suspect that it will not be a huge issue for us. 

Thanks again,

Josh

majenkotech, 2018-08-01:
There is no such thing as hardware PWM for servos in the Arduino API. The servo library just uses a timer to do a poor job of emulating pwm. SoftPWMServo does a better job with fewer resources because it is written to take advantage of the facilities of a more powerful platform.

majenkotech, 2018-08-01:
If you want to use hardware PWM you can just use the normal analog functions.  Such things as:

    analogWriteFrequency(50); // Set to 50 Hz
    analogWrite(25); // Set to ~10% duty (0-255)

You may want to tweak the resolution to give finer grained control of the speed of your ESC:

    analogWriteResolution(16); // 16-bit resolution
    analogWriteFrequency(50); // Set to 50 Hz
    analogWrite(6554); // Set to ~10% duty (0-65535)

Of course, some combinations of frequency and resolution are unobtainable, but at 50 Hz you should be able to get 16 bit quite happily.  That is, of course, assuming a 50 Hz based ESC...