chipKIT32/chipKIT-core #337: analogReadConversionComplete()' used but never defined [enabled by default]

JacobChrist, 2017-03-23:
Just noticed this warning when compiling non-blocking analogRead funcitons.  I'm not sure why I didn't notice it before nor am I sure what it means.  But it bug me.

In file included from C:\subversion\rockleyphotonics.com\lcm\arduino-lcm\hardware\chipkit-core\pic32\cores\pic32/WProgram.h:12:0,

                 from C:\subversion\rockleyphotonics.com\lcm\arduino-lcm\hardware\chipkit-core\pic32\cores\pic32/Arduino.h:4,

                 from C:\Users\JACOBC~1\AppData\Local\Temp\arduino_build_430017\sketch\analogReadNonBlockingSimple.ino.cpp:1:

C:\subversion\rockleyphotonics.com\lcm\arduino-lcm\hardware\chipkit-core\pic32\cores\pic32/wiring.h:152:17: warning: inline function 'uint32_t analogReadConversionComplete()' used but never defined [enabled by default]

 inline uint32_t analogReadConversionComplete();

                 ^

majenkotech, 2018-02-16:
It's because (I believe) you have an `inline` in a header file, but the "inlined" function in a C file.  If you want it `inline` then the code should be in the header file too (which also makes it static). Otherwise (which I'd suggest), remove the `inline` - you can't `inline` code that's in a separate TU, so it's pretty much pointless.

JacobChrist, 2018-02-19:
I was just trying to use inline in the hopes of not changing the timing of the original code (pre non-blocking code ) by turning in to functions.  Probably not a big issue since the PIC32 is so fast compared to ADC sampling speed.