chipKIT32/chipKIT-core #345: Compiler warning about serialEventRun in main.cpp

EmbeddedMan, 2017-05-08:
Here's the warning:

```
C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.4.1\cores\pic32\main.cpp: In function 'int main()':

C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.4.1\cores\pic32\main.cpp:64:27: warning: the address of 'void serialEventRun()' will always evaluate as 'true' [-Waddress]

         if (serialEventRun) serialEventRun();

                           ^

```

JacobChrist, 2017-05-08:
It seems to me that this warning is coming up because serialEventRun() is always defined.  Since its always defined I'm not sure why the if statement is even needed (I can see why you would do such a thing if there could be a condition where it was not defined, say a board with no serial ports that does not include HardwareSerial.h, but that doesn't seem to be the case).  I'm a little perplexed as to why checking for != NULL suppresses the warning but it could be that the compiler is just not smart enough to see that this will always evaluate to true as well?  Why not just drop the if statement and bring it back if we have have a board that doesn't include HardwareSerial.h?

majenkotech, 2017-05-08:
I am not sure exactly why it's there either, but the Arduino code has it, so I copied it. Verbatim.

I am sure it can probably be reduced down to just `serialEventRun();` safely.

JacobChrist, 2017-05-08:
There are probably some lower end Atmel chips that don't have serial ports.  Possibly this reason.

EmbeddedMan, 2017-05-08:
OK, I will amend my pull request.

JacobChrist, 2017-05-08:
Is there some syntax to put into the c code to suppress the warning.  Like to say, hey, we know this is always true, don't tell us about the warning?  Just so we can keep parity with Arduino?

EmbeddedMan, 2017-05-08:
I'm not sure with this compiler if you can do that or not. I guess I'd rather just make our code 'right', even though it's not exactly the same as Arduino's then. We've done that in a number of other places in the core too. Since there are no chipKIT boards without hardware UARTs, it makes sense to just remove the check.

EmbeddedMan, 2017-05-08:
If we did want to suppress the warning, we could do it like this:

```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Waddress"
        if (serialEventRun) serialEventRun();
#pragma GCC diagnostic pop

```

That compiles without warning.

majenkotech, 2017-05-08:
One thing I may have missed - serialEventRun() is meant to be defined as `weak`. It looks like I missed that one.

HardwareSerial.h:extern void serialEventRun(void) __attribute__((weak));

EmbeddedMan, 2017-05-08:
Well, that doesn't quite work. When I change the line in HardwareSerial.h to 

`extern void serialEventRun(void) attribute((weak));
`

I get the following error at compile:

```
In file included from C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.4.1\cores\pic32/WProgram.h:20:0,

                 from C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.4.1\cores\pic32/Arduino.h:4,

                 from C:\Users\bschmalz\AppData\Local\Temp\arduino_build_369131\sketch\FubarinoSD_DSPI_Demo.ino.cpp:1:

C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.4.1\cores\pic32/HardwareSerial.h:345:34: error: expected initializer before 'attribute'

 extern void serialEventRun(void) attribute((weak));

```

EmbeddedMan, 2017-05-08:
Oops. My bad. Fixing now .

EmbeddedMan, 2017-05-08:
Yup, that's all good now. I'll change the code to this. Nice find Matt.

EmbeddedMan, 2017-05-08:
Pull request merged.

closed 2017-05-08 by EmbeddedMan