chipKIT32/chipKIT-core #349: Add multi-SPI port example for DSPI library

EmbeddedMan, 2017-05-10:
It would be really nice to have some very simple DSPI examples, like one that just shows how to use multiple SPI ports, and another that shows how to use DSPI's interrupt feature.

EmbeddedMan, 2017-05-10:
I created two sketches (DSPI_Demo and DSPI_Int_Demo). They show using three hardware SPI ports on a Fubinaro SD. Here are the logic analyzer traces  for each.

This first one is the DSPI_Demo sketch. You can see three different transactions, each with three bytes, on three separate SPI ports. They each happen at a different speed as well. Click to embiggen.

[image]

And here is the DSPI_Int_Demo sketch output. Similar setup, but this time using SPI interrupts, so we can do other things while data is transferring in the background. Click to embiggen.

[image]

majenkotech, 2017-05-10:
Nice.

Can I put it a request for a "README.md" file to be added to new (and existing if people ever get time and a round tuit) examples? UECIDE now has an Examples Browser and will display it (as Markdown) when you select the example. Also it will (or should) show up in Github nicely.

Just a brief overview of what it is and how you use it will do.

EmbeddedMan, 2017-05-10:
Yeah! That's a great idea. I'll do that for these examples.

EmbeddedMan, 2017-05-12:
Pull request merged.

closed 2017-05-12 by EmbeddedMan