chipKIT32/chipKIT-core #365: File size incorrect

majenkotech, 2017-08-26:
https://github.com/chipKIT32/chipKIT-core/blob/239946479b0069d01410f53affa86b1174684551/package_chipkit_index.json#L89

The file size of 7674509 is incorrect. It should be 7674539.

EmbeddedMan, 2017-08-26:
> https://github.com/chipKIT32/chipKIT-core/blob/
> 239946479b0069d01410f53affa86b1174684551/package_chipkit_index.json#L89
>
> The file size of 7674509 is incorrect. It should be 7674539.

OK, should be fixed. I'll close when Rick reports it's working for him.

majenkotech, 2017-08-26:
Glad you noticed I linked to the wrong line number there - just checking you were awake ;)

Testing it now.

majenkotech, 2017-08-26:
That's fixed it for me. Install, uninstall, reinstall, success.

EmbeddedMan, 2017-08-26:
Fixed for me too. Thanks so much for finding the problem Matt!

majenkotech, 2017-08-26:
You'd think the IDE would check the size after download, wouldn't you...?

EmbeddedMan, 2017-08-28:
Rick has confirmed that this is now working for him.

closed 2017-08-28 by EmbeddedMan