chipKIT32/chipKIT-core #368: PIC32MZ support ?

Serveurperso, 2017-10-22:
Need a support for the bare minimum PIC32MZ board like the OLiMEX dev board.
To migrate my home project from PIC32MX (https://www.serveurperso.com/?page=robot).
I'm free for testing (with two OLiMEX boards)

Low level API wanted :

- UART with custom interrupt code and Pipe software FIFO from Matt
- Watchdog (lib from Matt)
- SPI / Wire (MPU9250)
- Hardware PWM (analogWrite)
- attachInterrupt for PIN RISING
- attachInterrupt for TIMER
- pinMode / digitalRead / digitalWrite / 8-bits direct ports access

Pascal

majenkotech, 2017-10-22:
What you are asking for is **not** PIC32MZ support (**which we already have**) but a specific board definition for your specific board.

Serveurperso, 2017-10-22:
It's not possible to do a generic board definition with maximum I/O and hardware init ?

Serveurperso, 2017-10-22:
I need to get all I/O offered by the PIC32MZ2048 on a generic board

majenkotech, 2017-10-22:
Which PIC32MZ2048? There's lots of them. All of them different, and all of them needing special settings.

Serveurperso, 2017-10-22:
I have the PIC32MZ2048EFG144

majenkotech, 2017-10-22:
Ok. That is step one. Now we know what number to enter into one of the settings. That's about 2% done. The second step is to work through schematics and datasheets for both the chip and the board and match up where peripherals need to be mapped to. Getting that right takes a lot of time and effort. Then definitions for that specific board can be created.

There is no such thing as a "generic" board with MZ - PPS makes that impossible.

Serveurperso, 2017-10-22:
https://www.olimex.com/Products/PIC/Development/PIC32-HMZ144/resources/PIC32-HMZ144_UM.pdf
I must learn how to do the definitions... the 6*UART part is already done, same for a lot of MZ I think:D

Serveurperso, 2017-10-22:
Now I understand "PIC32MZ has Peripheral Pin Selection hardware that has to be configured."