chipKIT32/chipKIT-core #369: I2C communication problems

parmiani, 2017-10-30:
Hello,
i have a chipkit wifire with PIC32MZ2048EFG100 mounted. I'm doig some communication tests with an IMU 10DOF with an MPU6050, an HMC5883 magnetometer and an MS5611 barometer. I'm using library written by Jeff Rowberg and I2C communication is managed by DTWI library. HMC5883 is connected to the aux bus of MPU. MPU6050 responds correctly while magnetometer does not respond. I have debugged the program checking with an oscilloscope on the aux bus and the I2C request is present also with the acknoledge of HMC5883,  but seems the microcontroller ingnores this and does not take the data. The address of MPU is 0x68 and that of HMC5883 is 0x1E.
Does anyone have some suggestion for me?

Thanks a lot

Alessandro

EmbeddedMan, 2017-10-30:
Alessandro,

I don't have any experience with any of the sensors you're using, so this
may not have anything to do with your problem at all. However, I'm going to
throw out something I just ran into on another project talking to a thermal
sensor over I2C.

I could never get the thermal sensor to send me back anything (or even ack)
when I tried to do a read of its registers. It turns out that the sensor
*required* a repeated start, rather than a full stop and full start in
between the write and read phases of the I2C transaction.

So I recommend that you check the datasheets of the sensors you are using
to see if any of them require this. I believe the DTWI library has the
ability to do repeated starts, but there is something special you have to
do to tell it to use them for reads.

If you have it, can you take a detailed logic analyzer capture of a failing
I2C transaction and post it so we can download it and look at it in detail?
That usually helps find the problem.

Thanks-

*Brian

parmiani, 2017-10-31:
Ok, thank you for the suggestion. 
I will post an I2C transaction

Alessandro

parmiani, 2017-11-01:
Hello,
these are the failing I2C transaction
[image]
[image]
[image]

parmiani, 2017-11-01:
Here there are the right I2C transaction done with another board:

[image]
[image]
[image]
[image]

These are the test of reading the 3 identification register of the chip (0x1E) with a burst read (0x0A the first, followed by the content of 0x0B and 0x0C).
At the end I was able to understand the problem.
There is a problem when i read the data from the sensor. When the master send for the second time the address of the slave for reading data, there is an error.
This address has to be sent with a 'read' bit (and not a write bit as in the picture); moreover, after that, the master resends for the third time the right address of the slave with 
the 'read' bit. In any case this causes an error and the magnetometer doesn't respond.
I solved going in the I2Cdev.cpp in function 'readBytes' and commenting the second Wire.beginTransmission(devAddr). Probably in the previous version of chipkit core
that uses twi library instead of the new one DTWI doesn't cause this problem.