chipKIT32/chipKIT-core #371: Wire.read() returns incorrect type and incorrect value when buffer is empty

bperrybap, 2017-11-19:
I mentioned this in issue #370 
The return value from Wire read() is the wrong type.
read()  should return an int vs a uint8_t
*AND* If there is no data in the buffer, -1 is returned.
So the lower eight bits of the int are the data value if the data value is >= 0
Every other core I've seen: Arduino AVR, Arduino sam, Arduino samd, Intel, esp8266, Teensy, Teensy3, Adafruit AVR, all work this way. 
The pic32 core Wire code returns a zero when there is no data in the buffer so it looks like a byte with a value of zero is available, when really there was a read error since no data was in the buffer.
The pic32 core is non compliant.

This is an easy fix and if there are no objections, I'll generate a PR for it.
--- bill

EmbeddedMan, 2017-11-19:
Bill, this will be a good fix to chipKIT's wire library. Nice find. We are
always looking for ways to become more compatible with what the other cores
do.

EmbeddedMan, 2018-03-24:
The PR has been merged so I'm closing the issue.

closed 2018-03-24 by EmbeddedMan