chipKIT32/chipKIT-core #387: All DFATFS formatted volumes get the same UUID

majenkotech, 2018-02-25:
When you format a filesystem within chipKIT using `DFATFS::fsmkfs(...)` the UUID assigned to the volume is always the same.  This value is based on a statically specified timestamp and never changes.

It's intended that it should get the time from an RTC and use that for the UUID - however not many chipKIT users have a board with a functioning RTCC.

A UUID is, as the name suggests, supposed to be a Universally Unique IDentifier. All ours are `0xBD690000` and are the result of:

```
#define _NORTC_MON  11
#define _NORTC_MDAY 9
#define _NORTC_YEAR 2014
#define GET_FATTIME()   ((uint32_t)_NORTC_YEAR << 25 | (uint32_t)_NORTC_MON << 21 | (uint32_t)_NORTC_MDAY << 16)
```

I would suggest using a random value would be better - however random values seldom are, especially if `srand()` hasn't been run with some source of entropy - so you'd get the same value showing up anyway. 

So ... what to do ...

The main thing is that on Linux you get /dev/disk/by-uuid symbolic links, such as:

```
lrwxrwxrwx 1 root root  9 Feb 25 17:02 BD69-0000 -> ../../sde
```

And if you plug in two chipKIT boards that have both formatted their own filesystems then that can't happen - you'd get one but not the other, and no way of knowing which it was you got.  

So this really is a perennial problem of microcontrollers ... how do you get a truly random number out of a predictable execution...

majenkotech, 2018-02-25:
Maybe I'm just over thinking things. Maybe I should just use `rand()` and leave it up to the user to properly use `srand()` with some entropy source. After all, on the MZ chips we already use the TRNG to properly seed the `rand()` function...

EmbeddedMan, 2018-02-25:
It feels like that's probably the best you can do. As long as there's an
easy way to over-ride the value from sketch (in case you want to have
control over your volume IDs).

*Brian

majenkotech, 2018-02-25:
Hmmm... that would mean changing the fsmkfs prototype to include a uint32 volume id value...

It could be done I guess, but would mess with the random generating, as you can't have a default value of a random number. Maybe with overloading it could work...

EmbeddedMan, 2018-02-25:
Maybe it's not that useful - who'd ever want to specify a GUID? Except
maybe for some strange testing . . .

majenkotech, 2018-02-25:
Doesn't hurt to add the option I guess. It should be simple enough.

majenkotech, 2018-02-25:
```
lrwxrwxrwx 1 root root  9 Feb 25 23:42 DEAD-BEEF -> ../../sde
```

:) I think that works nicely. Time for yet another PR...

majenkotech, 2018-02-25:
Fixed in #388

closed 2018-02-25 by majenkotech