chipKIT32/chipKIT-core #390: Breaking change for OpenBCI 32

biomurph, 2018-03-05:
This version of the chipkit core will successfully upload to the OpenBCI Cyton board (OpenBCI 32 in the boards menu). 
I checked this variants/openbci/Board_Defs.h against the one that I have that works, and it looks like on line 62  there is a #define _USB. Could that be causing the problem?
Right now, we are telling our customers to use a previous version of the core.

majenkotech, 2018-03-05:
What exactly is not working?

biomurph, 2018-03-05:
The board uses DSPI to communicate to devices on the board, and with the latest core there is no response from the devices. When I use the earlier version (1.4.3, 1.3.1) I don't have this problem.
It seems that the #define _USB is causing the problem? We do not use the USB on the OpenBCI 32.

majenkotech, 2018-03-05:
It is highly unlikely that USB will cause any problems. `_USB` is used to enable the new USB stack in the core. Nothing to do with DSPI.

majenkotech, 2018-03-05:
Setting the USB profile to "Custom" in the menu should have the same effect as disabling USB. It will still include the code, but it won't be actively doing anything. Only if you manually create the correct USB objects will anything be done with USB.  You can try that, but I don't think it'll make the slightest bit of difference.

majenkotech, 2018-03-05:
Hang on - are you using B10 or B11 for the SPI comms at all?  Those are the USB lines, so it may have an effect if you are.

majenkotech, 2018-03-05:
After decyphering your schematics it looks like yes, you are.

Ok, so 2 things to do: 

1. Remove `#define _USB` from the Board_Defs.h file
2. Remove the menu entries from the board.txt for the OpenBCI.

That should fix it.

biomurph, 2018-03-05:
Yes, I've done a compare on your latest version against my 1.4.3 and in the board.txt there are the lines
openbci.usb.enabled=true
openbci.usb.default=a_cdcacm

Which in addition to the define in the variants is jamming my SPI pins.
I will test to verify that's the only issue.

majenkotech, 2018-03-05:
Those two lines are only used by a perl script to generate the menu entries.  They are lower down the file:

```
openbci.menu.USB.a_cdcacm=Serial
openbci.menu.USB.a_cdcacm.build.extra_flags=-D__USB_ENABLED__ -D__USB_CDCACM__ -D__SERIAL_IS_USB__
openbci.menu.USB.b_cdcacm_km=Serial, Keyboard & Mouse
openbci.menu.USB.b_cdcacm_km.build.extra_flags=-D__USB_ENABLED__ -D__USB_CDCACM_KM__ -D__SERIAL_IS_USB__
openbci.menu.USB.z_custom=Custom / Disabled
openbci.menu.USB.z_custom.build.extra_flags=
```

majenkotech, 2018-03-05:
Just setting the profile to "Custom / Disabled" should prove it is the problem.  When I am back in Linux (I'm booted in Windows at the moment) I will make a PR for the changes.

majenkotech, 2018-03-05:
Since you're not using USB the MX150 would have been a better choice. Identical to the MX250 except it has no USB peripheral. Shave a few cents off the BOM cost...

biomurph, 2018-03-05:
I just tried your fix. Changed from 
openbci.menu.USB.z_custom=Custom / Disabled
to
openbci.menu.USB.z_custom=Disabled / Custom

Didn't do a bit of work. I'm going to try a couple of other things.

Regarding the MX150 v MX250, yeah, tell me about it. We want to make that change soon ;)

majenkotech, 2018-03-05:
I have put the changes in PR #391. You can try using that PR version to test it.

majenkotech, 2018-03-05:
"Custom / Disabled" is just the text that appears in the menu. You have to actively select it in the menu for it to do anything.

biomurph, 2018-03-05:
Got it
Stand by

biomurph, 2018-03-05:
OK, this is getting a little thorny...
When I select 'Custom / Disabled' from the menu my code won't compile because it throws an error 
'Serial0 is not a thing'
We are using the 2nd hardware UART , as well as the first hardware UART, and so that is not the right fix...

majenkotech, 2018-03-05:
Without USB Serial0 becomes Serial.  Before you had USB defined, but not turned on - that caused the Serial objects to change names.

I guess you could "fudge" it with 

    #define Serial0 Serial

in the Board_Defs.h file...

biomurph, 2018-03-05:
OK....

majenkotech, 2018-03-05:
Your original setup was a screwy mess that had USB enabled but not used (unless you went `Serial.begin(115200)`) which meant that the serial objects got renamed to fit the USB in. That should never have been. You should have had USB disabled from the start.

majenkotech, 2018-03-05:
I won't add that to the PR just yet - I'll wait for you to confirm it fixes it OK.

biomurph, 2018-03-05:
I totally get it. Our original setup was a screwy mess and it's pretty much been that way for years.
Our work with you and chipKIT is just the tip of the iceberg.

Selecting the Cutsom/Disabled option with the #define Serial0 Serial in the Board_Defs.h file will compile and upload, and I am able to communicate with my SPI devices. YAY! (added it on line 293, just where the _SER0 defines are)

I want to verify that Serial0 still works before I call this a total win. Should be able to verify soon, and we'll let you know via PR.

Thanks for your help Majenko, you have always rocked.

majenkotech, 2018-03-05:
Great.  It's already queued in PR #391 so you don't need to worry about making a PR.  Just let me know if there's anything else broken.

aj-ptw, 2018-03-05:
Just a reminder here Joel but please check to make sure this does not break `Serial1` as well?

majenkotech, 2018-03-05:
I can't see how it would. Serial1 remains serial1 always, regardless of the USB settings.

biomurph, 2018-03-06:
At least we're not getting a compile error on Serial1. Will test it tomorrow.

aj-ptw, 2018-03-07:
@majenkotech I can confirm that `Serial1` is not broken and is working as is.

@biomurph test completed

biomurph, 2018-03-07:
Sweet! Thanks @aj-ptw 
@majenkotech please proceed with the PR

majenkotech, 2018-03-07:
Excellent. I'll close this issue now, and hopefully @EmbeddedMan will work through the PR queue soon and we can release a new core version.

biomurph, 2018-03-07:
Great! We will await the new core release to update our users.
Always a pleasure to work with you and the chipKIT developers. 

Cheers!

closed 2018-03-07 by biomurph

EmbeddedMan, 2018-03-08:
Fantastic, We'll get a new core release out ASAP, probably this weekend.

EmbeddedMan, 2018-03-11:
chipKIT-core v2.0.2 has now been released. Could somebody from the OpenBCI team double check that the new version has the changes needed to fix this issue? Thanks-

aj-ptw, 2018-03-11:
Hi I will tomorrow! Thanks

aj-ptw, 2018-03-15:
v2.0.2 is not working for me with error:
```
In file included from /Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/pins_arduino.h:263:0,
                 from /Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/WProgram.h:9,
                 from /Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/Arduino.h:4,
                 from /var/folders/9c/sgws80yn6pv8m8scy2dyj1z40000gn/T/arduino_build_451509/sketch/DefaultBoard.ino.cpp:1:
/Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/variants/openbci/Board_Defs.h:52:17: error: conflicting declaration 'HardwareSerial Serial'
 #define Serial0 Serial
                 ^
/Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/HardwareSerial.h:188:39: note: in expansion of macro 'Serial0'
                 extern HardwareSerial Serial0;

                                       ^
In file included from /Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/WProgram.h:20:0,
                 from /Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/Arduino.h:4,
                 from /var/folders/9c/sgws80yn6pv8m8scy2dyj1z40000gn/T/arduino_build_451509/sketch/DefaultBoard.ino.cpp:1:
/Users/aj/Library/Arduino15/packages/chipKIT/hardware/pic32/2.0.2/cores/pic32/HardwareSerial.h:182:23: error: 'Serial' has a previous declaration as 'CDCACM Serial'
         extern CDCACM Serial;
```

majenkotech, 2018-03-15:
It seems you're still getting `__SERIAL_IS_USB__` defined somewhere.

majenkotech, 2018-03-15:
I see the problem... `genmenu.pl` wasn't honouring the true/false flag for USB.  Fixed #393.

aj-ptw, 2018-03-15:
Oh crud i forgot to switch the USB mode to Custom, can you change our variants somehow to select the custom/disabled by default?

majenkotech, 2018-03-15:
It shouln't even be giving you the option, but a bug in `genmenu.pl` didn't remove the option.  It's fixed in the next release.

EmbeddedMan, 2018-03-15:
I should be able to get a 2.0.3 out before tomorrow morning.

*Brian

majenkotech, 2018-03-15:
Don't rush - I have another enhancement to the USB stack I want to add.

EmbeddedMan, 2018-03-15:
Very good. I'll wait until it's all in. Just let me know.