chipKIT32/chipKIT-core #392: Serial0 & Serial1 not working on Fubarino Mini

JacobChrist, 2018-03-11:
I'm trying to get the following code running on a Fubarino Mini.  This code will echo back whatever is received from the USB serial if a wire is connected between Rx/Tx of either Serial0 or Serial1.

```
//Serial  Fubarino SD USB, Fubarino Mini USB
//Serial0 Fubarino SD tx  8 rx  9, Fubarino Mini tx 17 rx 18
//Serial1 Fubarino SD tx 28 rx 29, Fubarino Mini tx 26 rx 25
void setup()
{
  Serial.begin(115200);
  Serial0.begin(115200);
  Serial1.begin(115200);
}
void loop()
{
  if (Serial.available()>0)
  {
    Serial0.write(Serial.read());
    Serial1.write(Serial.read());
  }
  if (Serial0.available()>0)
  {
    Serial.write(Serial0.read());
  }
  if (Serial1.available()>0)
  {
    Serial.write(Serial1.read());
  }
}

```

I've tried with chipKIT-core 2.0.1 and 2.0.2 and both have the same results.  Serial0 seems to be non responsive at all and Serial1 is receiving something but it appears as if there is a bit rate mismatch (which seems really weird).  I haven't looked at any signals with a scope yet.

Jacob

majenkotech, 2018-03-11:
And what if you do just a "print" to the serial ports? Do they come out right?

EmbeddedMan, 2018-03-11:
Weird. I'll be able to dig into this later this afternoon Jacob. I'll try
to duplicate your issue.

No problems with either serial port on Fubarino SD?

*Brian

JacobChrist, 2018-03-11:
One problem with my code above, you should not read twice when receiving one character.  Code changed to this:

```
if (Serial.available()>0)
{
  char c = Serial.read();
  Serial0.write(c);
  Serial1.write(c);
}
```

Serial1.write works.  Serial.print prints out the ascii value in base 10

Serial0 is not working.

majenkotech, 2018-03-11:
Works fine in 2.0.1 on a Lenny as well (same chip as the FB Mini).

And I was about to mention the double reading ;)

majenkotech, 2018-03-11:
My FB Mini is built into one of my monitors controlling the back light at the moment so I can't actively test it.

One thing: you don't have the 50MHz (or 48MHz or whatever it is) bootloader installed on that board, do you?

JacobChrist, 2018-03-11:
Great question Matt... The board I'm using is really really really old.  I think Brian gave it to me at Masters many many many years ago.  Must have been about 4-5 years ago.  I'll try to test with a newer board.

majenkotech, 2018-03-11:
Or use a pickit to read the config bits, or re-flash the bootloader to be sure...

JacobChrist, 2018-03-11:
Arg... Its the bootloader... The new boards work fine!

closed 2018-03-11 by JacobChrist

majenkotech, 2018-03-11:
:)