chipKIT32/chipKIT-core #402: Incombatilibity whit nextion dispaly and chipkit (2.0.3) sd library

pgunzelmann, 2018-04-08:
Hello,

inside of sd Library of chipkit are following defintions:
boolean exists(char *filepath);
boolean mkdir(char *filepath);
boolean remove(char *filepath);
boolean rmdir(char *filepath);
if I compare with the adruino definitons (for adruino bards) i find
  // Methods to determine if the requested file path exists.
  boolean exists(const char *filepath);
  boolean exists(const String &filepath) { return exists(filepath.c_str()); }

  // Create the requested directory heirarchy--if intermediate directories
  // do not exist they will be created.
  boolean mkdir(const char *filepath);
  boolean mkdir(const String &filepath) { return mkdir(filepath.c_str()); }
  
  // Delete the file.
  boolean remove(const char *filepath);
  boolean remove(const String &filepath) { return remove(filepath.c_str()); }
  
  boolean rmdir(const char *filepath);
  boolean rmdir(const String &filepath) { return rmdir(filepath.c_str()); }
May you will change.

I run into this issue trying out to use nextion library in combination whit chip kit.

majenkotech, 2018-04-08:
The SD library is horrendously out of date and very nasty anyway. Where possible I recommend to use DFATFS instead, but I guess in this case you can't.

Since it's just missing `const` attributes it should be simple enough to fix though.

majenkotech, 2018-04-08:
Possible fix in #404

pgunzelmann, 2018-05-19:
Followup:
By using Boards Manager and installed chipKIT by chipKIT Community Version 2.0.3 the issue still exists.
How do i know at which time will the change be included at package_chipkit_index.json ? 
Thanks for supportig so far, a manual copy of the chaged sd files run excellent ,;)

majenkotech, 2018-05-19:
This change will be in the next released version. It takes time for that to happen.  There's other fixes we need to get out there too.