chipKIT32/chipKIT-core #406: digitalPinToInterrupt seems not to be defined for chipKIT MAX32

pgunzelmann, 2018-05-19:
Using the sketc:
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
digitalPinToInterrupt(3);
}
works for Mega 2560 but if i switch to chipKIT MAX32 following error occur:

C:\Users\ENTWPIC\AppData\Local\Temp\arduino_modified_sketch_234261\libeval.ino: In function 'void loop()':

libeval:11: error: 'digitalPinToInterrupt' was not declared in this scope

 digitalPinToInterrupt(3);

                        ^

exit status 255
'digitalPinToInterrupt' was not declared in this scope

pgunzelmann, 2018-05-19:
I expect its because digitalPinToInterrupt is not defined at 
chipKIT-core/pic32/variants/Max32/Board_Defs.h

majenkotech, 2018-05-19:
It is actually a functionality that is lacking from the core completely, and is something that crossed my mind last night whilst putting together a board configuration for a new board - we really need to implement digitalPinToInterrupt in the core.

majenkotech, 2018-05-19:
As it is most board definitions have defines for `PIN_INTx` to kind of map the other way (select an interrupt number and you can know the pin number from that), but it's not ideal.

majenkotech, 2018-05-19:
Actually, I may have come up with a better, more generic, method of implementing it that uses the `PIN_INTx` definitions.  Something that can be dropped in the core and, if needed, overridden by a board.

majenkotech, 2018-05-19:
Generically fixed in #407 - individual boards may need tweaks after testing.

majenkotech, 2018-05-19:
I just found the exact same code written by someone else in some of the board definitions - so I have removed it to prevent redefinition warnings.