chipKIT32/chipKIT-core #411: Install Jekyll instructions out of order

JacobChrist, 2018-05-25:
On the following documentation page: http://chipkit32.github.io/chipKIT-core/docs_template

which is in the repo here:

.../chipKIT-core/docs/docs_template.md

The instructions have you install Jekyll prior to cloning the repo.  The install instructions have suggest that Jekyll will "rad" (should be read) the Gemfile from the cloned repo (which is not possible because it has not been cloned yet)

Install Jekyll
To install Jekyll bundle will rad the contents of the Gemfile in the docs directory.

JacobChrist, 2018-05-25:
Additional changes / errors.  I download GitHub Desktop 1.2.0.  To install Jekyll I had to do the additional that is missing from the instructions:

1. Add the cloned chipKIT-core to the GitHub Desktop.
2. (optional) Set preferred shell by going to File->Options...->Advanced (tab) then selecting Shell from the drop down.
3. Open the shell by going to Repository->Open in "selected shell"

JacobChrist, 2018-05-25:
After trying all three shells the instructions to install Jekyll did not work.  Specifically the "bundle install" command fails.

EmbeddedMan, 2018-05-25:
I'm attempting to contact David Powell, who I believe wrote these instructions, so we can assign this issue to him in GH to fix. Thanks for testing it out Jacob.

JacobChrist, 2018-05-27:
The "bundle install" command assumes Ruby is installed on your computer.  This maybe the only step that is missing from the instructions.  If I have time to verify this I'll update the instructions and make a pull request.

JacobChrist, 2018-05-27:
I got Jekyll to work on my Windows 10 computer by doing the following:

1. Installed ruby from this site: https://rubyinstaller.org/downloads/
2. After ruby was installed I had to run the "gem install bundler" command to get the bundler.
3. I was able to now run the "bundle install" command in the docs directory.
4. Run "bundle exec jekyll serve"
5. Open the web site "localhost:4000"