chipKIT32/chipKIT-core #417: Wire.write compatibility with avr

JacobChrist, 2018-06-01:
I was playing with this library yesterday with 2.0.3 core. 

https://github.com/adafruit/Adafruit_TCS34725

The Wire.write call causes a complie error due to trying to pass a unit32_t to it. Possibly need to add some overloads to the library to get it to align with avr code.

To get it to work I just cast in the library to a uint8_t before the call.

JacobChrist, 2018-06-03:
Anyone have issue with adding these functions to Wire.cpp?

```cpp
int TwoWire::write(uint16_t data){ return(write((uint8_t) data)); }
int TwoWire::write(uint32_t data){ return(write((uint8_t) data)); }
```

I think this matches the spirit of the API as defined here: https://www.arduino.cc/en/Reference/WireWrite but I have not looked at the actual Arudino code so that I can keep changes black box.

I have tested and it compiles.  I can't test functionality until next Thursday since I left the TCS34725 board at school.

majenkotech, 2018-06-03:
You should only need the uint32_t one - the uin16_t can be promoted to int without any truncation.

majenkotech, 2018-06-03:
https://en.cppreference.com/w/cpp/language/implicit_conversion

JacobChrist, 2018-06-03:
Yes, seems to work with just the uint32_t.  I changed the call to this to force uint16_t:

```cpp
  Wire.write((uint16_t)(value & 0xFF));
```

Then changed Wire.cpp to just have the one call:
```cpp
  int TwoWire::write(uint32_t data){ return(write((uint8_t) data)); }
```

Should I commit and make a pull request or should I test functionally first?  Seems safe, but that's what I thought about the analogWrite() change last week and I was wrong.

majenkotech, 2018-06-03:
This is just a forced casting of larger types. Can't cause any harm (famous last words...)

EmbeddedMan, 2018-06-07:
Very good. Done.

closed 2018-06-07 by EmbeddedMan