chipKIT32/chipKIT-core #429: dtostrf convertig error

pgunzelmann, 2018-06-26:
example simplified to show only the problem:
```
void setup() {
  // put your setup code here, to run once:
  double d=1.04;
  Serial.begin(9600);
  Serial.println(String(d));
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Leads as output at serial using ATmega2560 Meaga to
1.04
but on chipKIT MAX32 to

1.40000

i tried to work arround useing the dtostrf function, but unfortunately the same occurance.

please try to simulate and may confirm or to reject

JacobChrist, 2018-06-26:
Nice find. Do you know if correct expected output is documented anywhere on the Arduino.cc? I have not attempted to confirm this yet.

majenkotech, 2018-06-26:
`dtostrf` is an avr-libc specific (non-standard) function. We emulate it (or try to), but it uses the same code as is used by `String`. (actually, `String` just calls `dtostrf`).

That simply uses `snprintf` to format the number into a string. 

`Print::print`, on the other hand uses its own internal `printFloat()` function. Maybe we should replace our `dtostrf` function with something based around that function for consistent results?

majenkotech, 2018-06-26:
Actually, looking at it, it's not quite that simple. I mis-read your values - 1.4000 != 1.04...!

So this looks like the formatting isn't working, and is dropping leading zeros from the fractional part (it splits it into integer and fractional components then prints "%d.%d" of them).  Maybe that second `%d` should be `%0d`...

pgunzelmann, 2018-06-26:
Yes this is exactly why i opened the issue, i written an interpollation function and had wondering why it is not correct. Do to String funktion also seem to use the dtostrf  function it should be improved.

@JacobChrist as majenkotech mentioned the return values are incorect, indepent from Arduino.cc

JacobChrist, 2018-06-26:
Ah, I missed that. Formatting is not the issue.

EmbeddedMan, 2018-06-29:
It would be great to have a fix for this. Anybody have time to submit a PR?

majenkotech, 2018-06-29:
I'd like to port dtostrf from avr-libc if possible.

EmbeddedMan, 2018-06-29:
Very good. Let me know if you have a PR that needs testing and I'll help
out. Thanks for working on this Matt.

majenkotech, 2018-07-01:
Unfortunately, `dtostrf` relies heavily on `__ftoa_engine`, which is written in AVR assembly. So that's a bit of a non-starter really. Shame. It would have been a good way to get 100% compatibility. 

However I have just come across [this implementation](https://github.com/adafruit/Adafruit_WICED_Arduino/blob/master/cores/maple/avr/dtostrf.c) which is in pure C, using sprintf.  Could be improved to use snprintf instead for greater security...  and maybe improved formatting.

majenkotech, 2018-07-01:
Ok, trimming is not needed, the Arduino defaults to 2 decimal places only, which threw me. Also snprintf needs an extra space in the width for NULL which dtostrf doesn't include. So with a bit of trial and error it's now formatting the same (ish).

EmbeddedMan, 2018-07-24:
@pgunzelmann @JacobChrist Matt just pushed up some new code that should completely fix this issue. Can either of you confirm that the changes fix the reported issue?

EmbeddedMan, 2018-08-27:
I haven't heard back from anybody that Matt's change fixed things for you, but I'm going to assume that it did, so I merged the change in. A new core version will be coming shortly which will include this change, so I'll close the issue then.