chipKIT32/chipKIT-core #430: DSDVOL is seriously broken...

majenkotech, 2018-06-26:
All the boards with DSDVOL macros in them (and hence the example with the library) show the constructor as:

    DSDVOL(<dspi>, <cs pin>)

When in actuality it's:

    DSDVOL(<dspi>, <pullup>)

You *cannot* pass the CS pin to the constructor. Instead, it uses the DSPI object's internal understanding of what the chip select pin is (um... how can the DSPI library magically know what you have wired a chip select connection up to...?).

Instead that whole pullup rubbish should just be dropped (SD cards have their own pullups internally, so it's pointless, and anyway, the CS pin is actively driven, so a pullup is irrelevant), and that constructor should accept the CS pin number. The reliance on the DSPI object's CS pin control should be erased.

EmbeddedMan, 2018-06-29:
Matt, it seems like this issue can be closed now that we merged your PR. Is there any more to do here?

majenkotech, 2018-06-29:
Yup, it can close now.

closed 2018-06-29 by majenkotech