chipKIT32/chipKIT-core #435: servo library has inaccurate times on non 80MHz CPUs

EmbeddedMan, 2018-09-09:
Or at least that's how it appears. The following sketch does not generate the correct times on a FB Mini:

```
#include <Servo.h>

Servo THRout;

void setup() {
  THRout.attach(0);
}

void loop() {
  delay(1000);
  THRout.write(1000);
  delay(1000);
  THRout.write(1500);
  delay(1000);
  THRout.write(2000);
}
```

It does work on boards running at 80MHz. My guess is that the servo library does not correctly take into account FOsc, although that's just a guess at this point. 

Digging into the library now.

majenkotech, 2018-09-09:
```
#define usToTicks(_us)  (((_us)*5)/4)
#define ticksToUs(_ticks) ((((unsigned)(_ticks))*4)/5)
```

Looks hardcoded to me...

EmbeddedMan, 2018-09-09:
Exactly.

closed 2019-04-09 by ricklon