chipKIT32/chipKIT-core #444: SD card issue with Web server example on Wi Fire ver C

alniju, 2018-12-23:
SD card cannot mount, error 3. I have changed the version of chipkit core from 2.0.6 to 1.4.3  and everything works now perfectly !

majenkotech, 2019-03-14:
You need to manually enable the internal pullup on the SDI line.

majenkotech, 2019-03-14:
Fixed in PR #450

lstandage, 2019-10-09:
I think there's another issue. The Board_Defs.h file for the Wi-Fire has the following line:

`#define DefineDSDVOL(vol, spi) DSDVOL vol(spi, 53)     // Create an DSDVOL object`

However, pin 52 is the CS line for the SD card.

majenkotech, 2019-10-09:
Yes, it should be 52.  The problem here is that the parameter was originally used not for the CS pin but to specify the MOSI (or maybe it was the MISO) in order to enable the internal pullup on the pin. However all the code comments and things referenced it as being the chip select pin. Since the chip select made more sense the pullup control was moved to a #define in the board definitions (or leave it to the user to enable manually, or better still, add a real pullup resistor), and the parameter was converted to be, as it should be, the chip select pin. 

Not all examples were changed to reflect that it seems.