chipKIT32/chipKIT-core #460: SoftwareSerial fails to compile for chipKIT WiFire

jg2562, 2020-05-21:
Hi,

I seem to be having an error using the following minimal script and I was not able to compile SoftwareSerial.cpp library for the chipKIT WiFire. 

```
#include <SoftwareSerial.h>

void setup() {

}

void loop() {
}
```

It appears to be an error in the struct as it gave back this error.

```
Arduino: 1.8.12 (Linux), Board: "chipKIT WiFire, Rev C and Newer (FPU), Custom / Disabled"

~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int, uint32_t)':
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:700:14: error: 'volatile union __CNCONAbits_t' has no member named 'SIDL'
   CNCONAbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:703:14: error: 'volatile union __CNCONBbits_t' has no member named 'SIDL'
   CNCONBbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:706:14: error: 'volatile union __CNCONCbits_t' has no member named 'SIDL'
   CNCONCbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:709:14: error: 'volatile union __CNCONDbits_t' has no member named 'SIDL'
   CNCONDbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:712:14: error: 'volatile union __CNCONEbits_t' has no member named 'SIDL'
   CNCONEbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:715:14: error: 'volatile union __CNCONFbits_t' has no member named 'SIDL'
   CNCONFbits.SIDL = 0;
              ^
~/.arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries/SoftwareSerial/SoftwareSerial.cpp:718:14: error: 'volatile union __CNCONGbits_t' has no member named 'SIDL'
   CNCONGbits.SIDL = 0;
              ^
exit status 255
Error compiling for board chipKIT WiFire.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

majenkotech, 2020-05-21:
Software serial does not work on MZ chips and never has done. But then, why would you want it on a chip with 6 hardware uarts?

jg2562, 2020-05-27:
Unfortunately, the chipKIT WiFire only wires out 2 of the hardware uarts from what we found. We need at least 3 for our current project. Are there plans to implement Software Serial for the MZ chipset?

JacobChrist, 2020-05-27:
Indeed it does.  Huh, you can make your own variant.  The PONTECH NoFire is
one I made for a board that doesn't have WiFi but uses the MZ chip.  Set it
up in Board_Defs.h in the "Serial Port Declarations" section.

Jacob

jg2562, 2020-05-27:
That is a consideration we are looking into, but it would be easier to read from software serial for now if that's an option.