chipKIT32/chipKIT-core #484: Apple Silicon Mac build support

mgilmore42, 2023-02-28:
Hello,

I was wondering about a possible Apple Silicon build. I have a project were the only computer I have access to is an Apple Sillicon Mac and will be compiling and uploading firmware to the chipKIT on an OpenBCI EEG board. I have the following questions for discussion:

1. Would adding support for the Apple Silicon be in alignment with the project goals?
2. Would it be a trivial matter to compile for Apple Silicon? (I see there is a linux arm)

I appreciate any input!

JacobChrist, 2023-02-28:
The chipKIT is kind of disheveled though there has been an effort by one person to get the compiler working on 64-bit Apple products I wouldn't rely on it.  The best solution would be to use boot-camp or parallels to run Window or Linux and compile in the virtual machine.

mgilmore42, 2023-03-01:
@JacobChrist Thanks for the suggestion! It works correctly when virtualization windows.

closed 2023-03-01 by mgilmore42